christoph2/pyxcp #18: Utilize construct's enum access feature

waszil, 2019-03-01:
Hi,
I'm wondering if it would be possible to introduce enum member access like this (let's use the CommModeBasic type):

```python
# new enum-based type definitions:
ByteOrder = Enum(BitsInteger(1), enum.IntEnum('ByteOrder', 'INTEL MOTOROLA', start=0))
AddressGranularity = Enum(BitsInteger(2), enum.IntEnum('AddressGranularity', 'BYTE WORD DWORD RESERVED', start=0))

# new CommModeBasic type definition:
CommModeBasic = BitStruct(
    "optional" / Flag,
    "slaveBlockMode" / Flag,
    Padding(3),
    "addressGranularity" / AddressGranularity ,
    "byteOrder" / ByteOrder
)
```

This would enable one to check values like this:

```python
cmb = CommModeBasic.parse(b'\x48')
print(cmb.byteOrder == ByteOrder.INTEL)
>> True
```

What do you think?

felixn, 2019-03-01:
I can't try it right now, but isn't this already possible?
E.g. CommModeBasic.byteOrder.INTEL

https://construct.readthedocs.io/en/latest/meta.html#refering-to-inlined-constructs

waszil, 2019-03-01:
It only works for Structs.
```python
AttributeError: 'Transformed' object has no attribute 'byteOrder'
```
Well, it is rather a construct issue/improvement I guess.

felixn, 2019-03-02:
I couldn't get it to work with the BitStructs either, so I was about to open an issue with construct when I saw you already did: https://github.com/construct/construct/issues/781
Good work, thanks! I added a small example.

But the only argument (that I could think of that) against moving the embedded enums to own types is that it clobbers the namespace in pyxcp.types a bit. But since those enums should be unique with unique names anyways, I don't think that's an issue.
Pullrequest: #20

felixn, 2021-12-16:
I think this could be closed, see #20

closed 2021-12-17 by christoph2