christoph2/pyxcp #19: Slave byteorder is not used in master

waszil, 2019-03-01:
I have a slave device with Motorola byte order, and all the master methods that use struct.pack/unpack, are hard-wired with '<' which means Intel. Therefore these do not work, for example setMta.

I suggest that the MasterBaseType should have slaveByteOrder and/or slaveByteOrderStructChar fields, and these should be updated when parsing the connect response:

```python
response = self.transport.request(types.Command.CONNECT, 0x00)
result = types.ConnectResponse.parse(response)
...
self.slaveByteOrder = result.commModeBasic.byteOrder
if self.slaveByteOrder == 'MOTOROLA':
    self.slaveByteOrderStructChar = '>'
else:
    self.slaveByteOrderStructChar = '<'
```
and after this, the master methods can use this:

```python
# in setMta:
addr = struct.pack("%sI" % self.slaveByteOrderStructChar, address)
...
```

Or am I missing something?
Thanks!

christoph2, 2019-03-02:
You're absolutely right! 
Considering byte-order from CONNECT_Res is a missing required feature. 
I will integrate it ASAP.

closed 2019-03-02 by christoph2

waszil, 2019-03-03:
Great, thanks!

commit e23b3c90 (2019-03-02): Consider byte-order (fixes #19)