christoph2/pyxcp #25: Byte-order is statically little-endian in types.py struct members

waszil, 2019-03-07:
Hi,
unfortunately my XCP-slave device is still big-endian, and what has been a problem with CONNECT and has been fixed, now I see that it is a problem throughout the whole types.py.
Every type defined by the construct.Struct that has multi-byte fields, are defined with little-endian types, and therefore parsed incorrectly from a bytestream response from a big-endian slave-device.
As I know, there is no possibility to define fields with endianness to be evaluated runtime with constrcut, so right now I can only think of that we should define all structs for BE and LE case as well, and it should be chosen in the master service functions depending on the slave device byte order.
Or is there a better solution maybe?

waszil, 2019-03-07:
Well, turns out, that construct supports this. I have commited to my fork a possible solution, but I'm not sure, that the storing of the slave device properties is an elegant solution this way:
f811e2f10cd1d6403469b928bb59075098ac7b5d
What do you think?

christoph2, 2019-03-07:
Meanwhile I've wrapped the responses to a 'Responses' class. 
I will also push it to a branch; but your solution seems to be more reasonable.

christoph2, 2019-03-07:
Here it is:
https://github.com/christoph2/pyxcp/tree/byteorder

felixn, 2019-03-07:
One more candidate - it's possible to pass arguments to the parse function at runtime:
https://construct.readthedocs.io/en/latest/meta.html#nesting-and-embedding

For example:
```
>>> GetIDResponse = Struct(
...     "mode" / Int8ul,
...     Padding(2),
...     "length" / IfThenElse(this._.byteOrder == ByteOrder.INTEL, Int32ul, Int32ub),
...     "identification" / If(this.mode == 1, Byte[this.length])
... )
>>> GetIDResponse.parse(b'\x01\x00\x00\x02\x00\x00\x00\x12\x34', byteOrder=ByteOrder.INTEL)
Container(mode=1, length=2, identification=ListContainer([18, 52]))
>>> GetIDResponse.parse(b'\x01\x00\x00\x00\x00\x00\x02\x12\x34', byteOrder=ByteOrder.MOTOROLA)
Container(mode=1, length=2, identification=ListContainer([18, 52]))
```

To me, that seems like the most "pythonic" solution.

waszil, 2019-03-08:
Oh, that's nice! I like this one. (I should have read the docs more)

christoph2, 2019-03-08:
I wasn't aware of this trick - - accepted!

waszil, 2019-03-08:
I have created a pull-request https://github.com/christoph2/pyxcp/pull/26#issue-259441953 , containing the byte-order changes suggested by @felixn, and it also contains some new request/response types for DAQ .

waszil, 2019-03-08:
Actually, I have added the dataclasses module, but I see it can be a problem as it is only included as standard library from python 3.6. Maybe the SlaveProperties should only be a simple class...

closed 2019-03-08 by waszil