christoph2/pyxcp #30: download commands length calculation bug

waszil, 2019-05-28:
The download and downloadNext implementation calculates the length of the payload incorrectly:

    len(data)
    # correct would be:
    len(*data)

waszil, 2019-05-28:
this is true for base.py, pre35.py and py35.py, in all locations, where the function parameter is *data.

waszil, 2019-05-30:
One more thing: for block mode, as specified in the Protocol Layer Specification, the length parameter in DOWNLOAD and DOWNLOAD_NEXT packets has to be set to the whole lenght of the block, so this length calculation should not be done here, or at least there should be an optional parameter for the higher level block-mode implementation to be able to set the correct length in the packets.

felixn, 2019-08-22:
Hi waszil, I'm looking into this but can't really find what you mean.
I think you're talking about the lines `length = len(data)` in download() and downloadNext() in pyxcp/master/base.py. They are tested by the methods testDownload and testDownloadNext in pyxcp/tests/test_master.py, and the length field is checked to be 4 for both cases. I think for byte granularity that is the correct value? There might still be an issue for different granularities...

waszil, 2019-08-23:
Hi, my apologies, you are correct, I have totally misused these... I see, in the tests these methods are called like this: download(*data), I used download(data), and there is the problem... Sorry again.
However, I wonder, why is it necessary to use *data in the method's argument? What is gained via this starred unpacking of the data parameter? In my opinion it would be much cleaner just to use simply 'data' as is, as it is already typed to be bytes.

christoph2, 2019-08-23:
Well, it's basically my stupid design fault -- variadic methods are completely nonsensical in this context,  I'll fix the signatures.

P. S. : welcome back guys :-)

commit e5ccc1d7 (2019-08-23): Pass arguments as lists -- fixes #30
closed 2019-08-23 by christoph2