christoph2/pyxcp #54: Unable to connect after disconnect() in base.py

Lecerof, 2020-08-05:
The helloxcp.py example works fine, but just once. When the script is run for a second time I get

[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
ERROR:pyxcp.pyxcp.master.errorhandler:XcpTimeoutError

Seems like it does not close the connection entirely, but I am not sure.

christoph2, 2020-08-05:
I've never observed such behaviour --chances are,  your XCP slave isn't working correctly. 
Could you give some details (XCP slave,  transport)? 
You should also add
```-l DEBUG```
to your command line and paste the trace output here.

Lecerof, 2020-08-05:
Yes, you might be correct. Not sure I can give so many details about the XCP slave, anything specific you have in mind? I am trying to connect over ethernet with UDP protocol.

The debug information yields the following on the first attempt:

```
DEBUG:pyxcp.transport.Base:CONNECT
DEBUG:pyxcp.transport.Base:-> [02 00 00 00 ff 00]
DEBUG:pyxcp.transport.Base:<- L8 C0 [ff 01 00 08 08 00 01 01]
DEBUG:pyxcp.transport.Base:GET_ID
DEBUG:pyxcp.transport.Base:-> [02 00 01 00 fa 01]
DEBUG:pyxcp.transport.Base:<- L8 C1 [ff 00 00 08 00 00 00 00]
DEBUG:pyxcp.transport.Base:SET_MTA
DEBUG:pyxcp.transport.Base:-> [08 00 02 00 f6 00 00 00 58 ef 06 40]
DEBUG:pyxcp.transport.Base:<- L1 C2 [ff]
DEBUG:pyxcp.transport.Base:UPLOAD
DEBUG:pyxcp.transport.Base:-> [02 00 03 00 f5 04]
DEBUG:pyxcp.transport.Base:<- L5 C3 [ff 42 20 00 00]
DEBUG:pyxcp.transport.Base:DISCONNECT
DEBUG:pyxcp.transport.Base:-> [01 00 04 00 fe]

Slave properties:
=================
ID: ''
{'addressGranularity': EnumIntegerString.new(0, 'BYTE'),
 'byteOrder': EnumIntegerString.new(0, 'INTEL'),
 'maxCto': 8,
 'maxDto': 8,
 'maxWriteDaqMultipleElements': 0,
 'optionalCommMode': False,
 'protocolLayerVersion': 1,
 'slaveBlockMode': False,
 'supportsCalpag': True,
 'supportsDaq': False,
 'supportsPgm': False,
 'supportsStim': False,
 'transportLayerVersion': 1}
DEBUG:pyxcp.transport.Base:<- L1 C4 [ff]
```

And on the second attempt:

```
DEBUG:pyxcp.transport.Base:CONNECT
DEBUG:pyxcp.transport.Base:-> [02 00 00 00 ff 00]
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
ERROR:pyxcp.pyxcp.master.errorhandler:XcpTimeoutError
DEBUG:pyxcp.transport.Base:CONNECT
DEBUG:pyxcp.transport.Base:-> [02 00 01 00 ff 00]
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
ERROR:pyxcp.pyxcp.master.errorhandler:XcpTimeoutError
DEBUG:pyxcp.transport.Base:CONNECT
DEBUG:pyxcp.transport.Base:-> [02 00 02 00 ff 00]
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
ERROR:pyxcp.pyxcp.master.errorhandler:XcpTimeoutError

... 

DEBUG:pyxcp.transport.Base:-> [02 00 32 01 ff 00]
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpTimeoutError
ERROR:pyxcp.pyxcp.master.errorhandler:XcpTimeoutError

...

And continues ticking for a while. Don't know the last entry.

```

Thank you so much for your advice.

christoph2, 2020-08-05:
The first attempt looks like it should look, no errors,  disconnect()  accepted. 
At this point the internal state-machine (or whatsoever) of the XCP slave should make a transition from connected to disconnected state, waiting for a new connect() request. 

Really looks like a. minimalistic , homebrewn XCP implementation (formerly used in a CAN project?  there's no reason to restrict payload size to 8 on Eth) ...