christoph2/pyxcp #60: drop python 3.4 support

danielhrisca, 2020-11-16:
see https://www.python.org/dev/peps/pep-0429/

christoph2, 2020-11-20:
Yep,  I want to get rid of this pre35/py35 stuff in any case,  but for some reason I still need 3.4 support for the next few months. 

Regarding logging: 
I think application logging and bus logging should be separated. 
Bus-loggers should be pluggable / configurable -- one promising candidate are PCAP files (*scapy*  could be used on the Python side) 
[xcpdump ](https://github.com/christoph2/asamint/blob/master/tools/xcpdump/xcpdissect.c) includes decent decoding logic, it would be easy to create a standalone program with no SocketCAN dependencies.

felixn, 2021-12-16:
This is done now (/ we did it a while ago) :)
#64

closed 2021-12-17 by christoph2