christoph2/pyxcp #83: pyxcp/transport/can.py listen function should check if len of frame.data is bigger than 0

nedlukasz, 2021-11-26:
There is possible that `frame = self.canInterface.read()` will return frame with len(frame.data) == 0. Frame with empty data will cause exception inside processResponse in code `pid = response[0]`.

Such case when Can Driver return frame with empty data exist for Kvaser Virtual Can Driver.

danielhrisca, 2021-11-26:
The XCP frame must have at least the first data byte, so there is some other problem

nedlukasz, 2021-11-26:
Yes, XCP frame must have at least one byte. Maybe better place for that check is inside pyxcp/transport/candriver/python_can.py. There is already in line 112 check about correctness of frame
`if frame is None or frame.arbitration_id != self.parent.can_id_master.id:
                return None `
and it can be extended as:
`if frame is None or frame.arbitration_id != self.parent.can_id_master.id or not len(frame.data):
                return None`

commit 11776fa3 (2021-12-08): Fixes #83
closed 2021-12-08 by christoph2