christoph2/pyxcp #85: For x64 based machines no option to use x64 seed and key dll

still-learnin, 2021-12-07:
On an x64 based Windows machine even if an x64 seed and key dll is available the package still forces the use of an x32 dll and the bridging program (at least I see no other option). It would be good if there was a configuration flag to allow the possibility of using an x64 dll. It certainly works on my system if I force the use_ctypes branch in getKey().

Is there any possibility to make this configurable.

btw. Thanks for making this project available it is extremely useful.

christoph2, 2022-01-11:
There are now two new configuration parameters:

- SEED_N_KEY_DLL
- SEED_N_KEY_DLL_SAME_BIT_WITH, basically setting use_ctypes to True (haven't found the time to test it yet)

still-learnin, 2022-01-11:
I have tried this and it certainly works well for me.

I was slightly surprised to see that you moved the configuration of the dll from a method to the 'config' as well. Mainly because it would break backwards compatibility but also it is conceivable that one could reprogram the slave firmware, at which point the master might need to use a different seed_n_key file. In practice a single product is likely going to be using the same seed_n_key algorithm between different firmware releases - certainly this is my use case.

This could be circumvented by tearing down the current XCP master and creating a new one, however I must say that, when  I tried this a while ago to get around some other issues I was having, I didn't have much success, with getting everything cleaned up and deleted nicely.

christoph2, 2022-01-12:
The attribute **seedNKeyDLL** is still accessible, so there is (should) no code breaking here.

still-learnin, 2022-01-12:
my bad....yes of course, for some reason in my mind I thought it was a nominally protected attribute (although of course still accessible).

closed 2022-01-12 by still-learnin