christoph2/pyxcp #91: _block_downloader needs to consider case where "remaining" branch not called

still-learnin, 2022-01-18:
I didn't spot this in my original example but there is the use case for the _block_downloader where the data fits exactly in an integer number of packets. In this case the "remaining" branch is not called and the expected response is not waited for. So the calls to the dl_next and dl_next_func in the main for loop also need to consider setting the 'last' flag, something like:

        for index in packets:
            packet_data = data[offset : offset + max_packet_size]
            last = (remaining_block_size - max_packet_size) == 0:
            if index == 0:
                dl_func(packet_data, length, last)  # Transmit the complete length in the first CTO.
            else:
                dl_next_func(packet_data, remaining_block_size, last)
            offset += max_packet_size
            remaining_block_size -= max_packet_size
            delay(minSt)

christoph2, 2022-01-18:
OK, fixed.

P.S.: Many thanks for your valuable contributions, but couldn't you just fork and PR?
This would make diffing, merging, and the like, much easier for all involved...

still-learnin, 2022-01-18:
Sorry, yes this is the first open source package I have used in anger and so didn't really know the process (was an embedded software engineer for many years and have just returned after an 8 year break - Github wasn't really much of a thing then as far as I can remember).

danielhrisca, 2022-01-18:
> I have used in anger

I had to google that one to get the real meaning 😄

closed 2026-02-01 by christoph2