christoph2/pyxcp #92: pypi still only shows version 0.18.1 avaialable

still-learnin, 2022-01-19:
Is there a reason that pypi shows the latest release as 0.18.1? I have never noticed a delay before once a release has been made on GitHub. But maybe this is intentional?

christoph2, 2022-01-19:
No, that's not intentionally;
 I'm trying to incorporate new functionality (a recorder for XCP frames (mainly measurements)), but the GitHub actions build works not as expected, which prevents upload to PyPI...
Sorry for your inconveniences, I'm working hard to fix it.

christoph2, 2022-01-19:
0.18.15 is now available on PyPI, versions 3.5 to 3.10, 
currently only **win_amd64**
Other platforms, esp. Linux are still WIP.

danielhrisca, 2023-12-13:
The legacy upload mode is not working directly now since the latest PyPI updates 

https://github.com/christoph2/pyxcp/actions/runs/7192759486/job/19590169114

danielhrisca, 2023-12-13:
For reference this is my latest pypi.rc file for uploading the asammdf wheels

```ini
[distutils]
  index-servers =
    pypi
    asammdf

[pypi]
  username = __token__
  password = pypi-<your_tocken_here>
  
[asammdf]
  repository = https://upload.pypi.org/legacy/
  username = __token__
  password = pypi-<your_tocken_here>
```

christoph2, 2023-12-13:
I've created a new PYPI API-key, the old one was probably invalidated after I switched to 2FA.