christoph2/pyxcp #93: pyxcp license

danielhrisca, 2022-03-01:
Hello Christoph,

as you probably know `GPLv2` is incompatible with `GPLv3` and `LGPLv3`. This make it impossible to incorporate pyxcp in an application together with other libraries. (see https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility)

More so in the CAN transport `pyxcp` imports `python-can` which is licensed under `LGPLv3` (see https://github.com/christoph2/pyxcp/blob/2cfbc6b3028cebc46f3a5a316ca713b6be4b29e7/pyxcp/transport/candriver/python_can.py#L32)

Would you consider switching to a more generally compatible license like `GPLv2 or later` or `GPLv3` ?

christoph2, 2022-03-01:
I've just replaced the LICENSE file with the official LGPLv3 text -- not sure if I also need to update the __copyright__ stuff...🙄

danielhrisca, 2022-03-01:
Thank you for the update.

> not sure if I also need to update the **copyright** stuff.

Now that you mention I see that the copyright notice has a reference to `GPL v2 or later`, so it was anyway not matching the previous license file. You probably need to update it, maybe you can just make a reference to the new license file

waynebeaton, 2022-03-15:
> I've just replaced the LICENSE file with the official LGPLv3 text -- not sure if I also need to update the **copyright** stuff...roll_eyes

@christoph2, if by "copyright stuff", you mean the file headers, then strictly speaking the answer is yes: you should also update the file headers to reflect the change of license. It's probably also a good idea to confirm with all of the copyright holders (i.e., the contributors) that they agree to the license change.

For consistency, at a minimum, I recommend that you at least change the license mention in the README file.

In lieu of a wholesale update of the headers, can you confirm that it is your intention that all of the files with copyrights attributed to you that currently show GPL-2.0-or-later are as of this commit being distributed under the LGPL-3.0-or-later? (while I believe that it's reasonably implied by https://github.com/christoph2/pyxcp/issues/93#issuecomment-1055510128 confirmation will mitigate potential for doubt).

christoph2, 2022-03-17:
Yep, I think LGPLv3-or-later is the right way to go.
I'm currently modernizing all of my projects, including installation and meta-data. Here's a related PEP:
https://peps.python.org/pep-0639/

danielhrisca, 2022-04-28:
Hello Christoph,

It seems that there are still files that contain references to the GPL license (for example https://github.com/christoph2/pyxcp/blob/d82c62e4a1fa799b9184a534afc44b0db516e4d5/pyxcp/cxx/exceptions.hpp#L10)

christoph2, 2022-04-28:
You're right Daniel,

I'll fix the C++ files today.

commit d614b8ef (2022-04-28): Remove copyright notice from C++ files. fixes #93
closed 2022-04-28 by christoph2