christoph2/pyxcp #95: pip install failed

shinhwap, 2022-03-11:
Hello, 
I'm trying install pyxcp by pip install.
But it will be failed at Preparing metadata (pyproject.toml)
here is the error log
```
Collecting pyxcp
  Using cached pyxcp-0.18.41.tar.gz (123 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Installing backend dependencies: started
  Installing backend dependencies: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'

  error: subprocess-exited-with-error
  
  Preparing metadata (pyproject.toml) did not run successfully.
  exit code: 1
  
  [20 lines of output]
  Traceback (most recent call last):
    File "E:\etas\venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 363, in <module>
      main()
    File "E:\etas\venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "E:\etas\venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 164, in prepare_metadata_for_build_wheel
      return hook(metadata_directory, config_settings)
    File "C:\Users\A50044\AppData\Local\Temp\pip-build-env-wg9qxrkl\overlay\Lib\site-packages\pdm\pep517\api.py", line 59, in prepare_metadata_for_build_wheel
      return _prepare_metadata(builder, metadata_directory)
    File "C:\Users\A50044\AppData\Local\Temp\pip-build-env-wg9qxrkl\overlay\Lib\site-packages\pdm\pep517\api.py", line 39, in _prepare_metadata
      dist_info = Path(metadata_directory, builder.dist_info_name)
    File "C:\Users\A50044\AppData\Local\Temp\pip-build-env-wg9qxrkl\overlay\Lib\site-packages\pdm\pep517\wheel.py", line 128, in dist_info_name
      name = self.meta.project_filename
    File "C:\Users\A50044\AppData\Local\Temp\pip-build-env-wg9qxrkl\overlay\Lib\site-packages\pdm\pep517\base.py", line 144, in meta
      self._meta.validate(True)
    File "C:\Users\A50044\AppData\Local\Temp\pip-build-env-wg9qxrkl\overlay\Lib\site-packages\pdm\pep517\metadata.py", line 87, in validate
      return validate_pep621(self._metadata, raising)
    File "C:\Users\A50044\AppData\Local\Temp\pip-build-env-wg9qxrkl\overlay\Lib\site-packages\pdm\pep517\validator.py", line 85, in validate_pep621
      raise PEP621ValidationError(validator.errors)
  pdm.pep517.exceptions.PEP621ValidationError: {'license': ['none or more than one rule validate', {'oneof definition 0': ['must be of dict type'], 'oneof definition 1': ['must be of dict type']}]}
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

Encountered error while generating package metadata.

See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```
I've tried python 3.7.4 64bit、3.7.4 32 bit、 3.9.10 64bit but all have the same error.
Can you check out what's the problem?

Thank you

still-learnin, 2022-03-16:
Did you find a solution to this problem?

I am currently on 0.18.15 and tried to upgrade to the latest version and found exactly the problem you have above. I have upgraded a number times with older versions and never had a problem. It seems the installation process must have changed. I did look on the internet for potential solutions, there is some talk of similar issues but its not really an area I have much experience in, and nothing seemed that trivial.

christoph2, 2022-03-17:
Ok, I hope it works for now.
And yes, the new-school build is still WIP.

still-learnin, 2022-03-22:
I have just tried this again but I still get an issue:

      pyxcp/recorder/lz4.cpp(117): fatal error C1083: Cannot open include file: 'lz4.h': No such file or directory

I also notice an error earlier on in the process (not sure if this really a problem though):

      running asamkeydll                                                                            
      Building pyxcp/asamkeydll.exe failed: '[WinError 2] The system cannot find the file specified'

I have not used or needed to build this dll, since I have 64 bit seed and key dll's already.

shinhwap, 2022-10-31:
Hello,
This problem had been solved from new version.

closed 2022-10-31 by shinhwap