christoph2/pyxcp #98: can interfaces not working

oliverbl, 2022-03-30:
version: 0.18.46

I can`t initialize any can interface, the can transport module does not find any sublasses and returns an empty list
```python
def registered_drivers():
    """
    Returns
    -------
    dict (name, class)
        Dictionary containing CAN driver names and classes of all
        available drivers (pyxcp supplied and user-defined).
    """
    sub_classes = CanInterfaceBase.__subclasses__()
    return OrderedDict(zip(([c.__name__ for c in sub_classes]), sub_classes))

````

I also noticed other issues:
```python
def isExtendedIdentifier(identifier: int) -> bool:
    """Check for extendend CAN identifier.

    Parameters
    ----------
    identifier: int

    Returns
    -------
    bool
    """
    return (identifier & CAN_EXTENDED_ID) == CAN_EXTENDED_ID
```
IDs can be extended although they are smaller than 11bit
I think the only way to be sure is to use an additional config variable to set extended or normal IDs.

Also the SLAVE and MASTER ID seem to be used reversed:
```python
    def transmit(self, payload):
        frame = Message(
            arbitration_id=self.parent.can_id_slave.id,
            is_extended_id=True if self.parent.can_id_slave.is_extended else False,
            is_fd=self.is_fd,
            data=payload,
        )
        self.bus.send(frame)
```
I think this should be the master ID

oliverbl, 2022-03-30:
The first issue can be resolved by importing the driver in the user code, for example:
```python
import pyxcp.transport.candriver.pc_pcan
```
Is this how it is supposed to work? Is there documentation about this?

oliverbl, 2022-03-30:
Ok the second issue can be solved by combining the IDs with the extended bit:  0x80000000

christoph2, 2022-03-31:
The first one was (now fixed) an installation issue, the required package [python-can](https://github.com/hardbyte/python-can) was missing.
Just run `pip install python-can`

`isExtendedIdentifier` works correctly, `CAN_EXTENDED_ID` **IS** defined as `0x80000000`, i.e. Autosar-style.

christoph2, 2022-03-31:
I've just added a tiny script -- [pyxcp-probe-can-drivers](https://github.com/christoph2/pyxcp/blob/master/pyxcp/scripts/pyxcp_probe_can_drivers.py) -- giving you an overview whats on your system.