christoph2/pyxcp #104: Push in master.py is failing due to incorrect interface

dkuschmierz, 2022-06-30:
pyxcp version affected: 0.18.52

While using the push function of master, the following problem occurs:

` File "xxxx\lib\site-packages\pyxcp\master\master.py", line 605, in push
    callback=callback,
  File "xxxx\lib\site-packages\pyxcp\master\master.py", line 676, in _generalized_downloader
    block_downloader(block)
  File "xxxx\lib\site-packages\pyxcp\master\master.py", line 729, in _block_downloader
    dl_func(packet_data, remaining, last=True)
  File "xxxx\lib\site-packages\pyxcp\master\errorhandler.py", line 376, in inner
    res = executor(inst, func, arguments)
  File "xxxx\lib\site-packages\pyxcp\master\errorhandler.py", line 337, in __call__
    raise UnrecoverableError("Don't know how to handle exception '{}'".format(repr(e))) from e
pyxcp.master.errorhandler.UnrecoverableError: Don't know how to handle exception 'TypeError("download() got an unexpected keyword argument 'last'")'
`

I think the download() function in master.py is missing the _last_ attribute (see program).

dkuschmierz, 2022-06-30:
Created PR:
https://github.com/christoph2/pyxcp/pull/105

christoph2, 2022-07-03:
Fixes issue #104

closed 2022-07-03 by christoph2