christoph2/pyxcp #110: Memory usage increases over time

ttlorenz, 2022-08-04:
It appears that XCP is increasing memory usage over time.

I actually could find at least a couple of spots where allocation happens while never freeing:
- self.evQueue.append(response) (line 292 of base.py, version 0.18.48)
- self.servQueue.append(response) (line 295 of base.py, version 0.18.48)
- self.daqQueue.append (line 314 of base.py, version 0.18.48)

As far as I understand these queues are mostly / only used for testing only. Is that the case?

As a workaround in my application I can definitely deque them from time to time but it would be nicer to handle that internally / enable them only for debug.

ttlorenz, 2022-08-04:
version 0.18.48

danielhrisca, 2022-08-04:
> * self.evQueue.append(response) (line 292 of base.py)
> * self.servQueue.append(response) (line 295 of base.py,)
> * self.daqQueue.append (line 314 of base.py)

In the current implementation you should consume those queues. All the ODTs are placed in the `daqQueue`

ttlorenz, 2022-08-04:
I see, thanks for the prompt answer!

Do you think this is going to be changed in the future?

danielhrisca, 2022-08-04:
I don't think so

christoph2, 2022-08-08:
Currently work in progress -- there'll be a much more flexible callback/callout mechanism in the (near) future.
The rekorder extension is one step in this direction (unfortunatly, there are still some C++ - Python integration issues).

ttlorenz, 2022-08-08:
Thank you for the answers!
Have a nice day