christoph2/pyxcp #114: Any OEM ECUs with open XCP?

rusefillc, 2022-08-31:
Sorry for an offtopic question again but what kind of hardware+A2L combo can I get on a bench to see pyxcp traffic?

danielhrisca, 2022-08-31:
Have a look at this https://github.com/vectorgrp/XCPlite

rusefillc, 2022-08-31:
@danielhrisca https://github.com/vectorgrp/XCPlite is amazing piece of software but my question is different. My question is what is the chepest (used?) _hardware_ with XCP already in it for me to experience the complete solution on the bench?

rusefillc, 2022-08-31:
@danielhrisca i.e. what do people usually use pyxcp for? What do you use pyxcp for, what device do you connect to?

danielhrisca, 2022-08-31:
Many of today's automotive ECUs support XCP.

You could modify the vector demo code and run it on a raspberry pi for example

rusefillc, 2022-08-31:
@danielhrisca do you have specific example of an automotive ECU for which you have A2L file?

hornkorb, 2022-11-22:
@rusefillc you won't get a a2l file from an automotive ECU. That file keep so many secret information.