christoph2/pyxcp #130: SLAVE_ID vs MASTER_ID

BeKl92, 2023-02-15:
Is it possible that slave_id and master_id are swapped within the code?

From my understanding the master_id should be the id pyxcp should use to send data to the ecu. The ecu should response with the slave_id. Other xcp implementations used by vector canape or by peakcan xcp driver interpret the master_id as the id used to send data to the ecu and the slave_id used to receive data from the ecu. 

So to summarize it according to xcp documentation: 
There are at least 2 ids one id for RES/ERR/EV/SERV/DAQ and another one for CMD/STIM.
The id used by the tool to send data is  CMD/STIM which should represent the master_id. 
The id used by the tool to receive data is RES/ERR/EV/SERV/DAQ which should represent the slave_id.

ttlorenz, 2023-02-15:
Hi,

what you are reporting is also something we have found within my development team.
At the moment, we are work-arounding the issue by simply swapping the meaning in our "manager" layer / config file.
But yes, code shall be fixed or at least documented.

cheers

BeKl92, 2023-02-15:
Thanks for confirming it. 
Yes i have also swapped it within the config file.

neresdu, 2023-06-28:
Yes for me is also swapped it, i've change in the config file and works