christoph2/pyxcp #137: XcpResponseError [ERR_ACCESS_LOCKED] -  Seed and Key unlock process error

neresdu, 2023-07-10:
hello everyone I would like to share the error I am having in the seed and key unlock process, for some reason the ECU does not unlock any features, I validated that my Seed and Key file is working using a third party software such as INCA software and it worked . Now the same process using pyXCP I am having the following error when sending the unlock command:
XcpResponseError [ERR_ACCESS_LOCKED
L2 C4 [fe 25]

I am using a Debian 10 and .so seed and key file

Code I'm using to unlock the ECU
```
from pyxcp.cmdline import ArgumentParser
from pyxcp import types

def callout(master, args):
    if args.sk_dll:
        master.seedNKeyDLL = args.sk_dll

ap = ArgumentParser(callout)
ap.parser.add_argument(
    "-s",
    "--sk-dll",
    dest="sk_dll",
    help="Seed-and-Key .DLL name",
    type=str,
    default=None,
)

with ap.run() as x:
    x.connect()
    rps = x.getCurrentProtectionStatus()
    print("Protection before unlocking:", rps, end="\n\n")
    print(x.getStatus())

    result = x.cond_unlock("daq") ## ERROR HERE
    print("X.cond_unlock result: ", result)
    rps = x.getCurrentProtectionStatus()
    print("Protection after unlocking:", rps)

    x.disconnect()
```

Here is the error log from pyXCP

```
DEBUG:pyxcp.transport.Base:Python-CAN driver: socketcan - socketcan channel 'can1']
DEBUG:pyxcp.transport.Base:CONNECT
DEBUG:pyxcp.transport.Base:-> [ff 00]
DEBUG:pyxcp.transport.Base:<- L8 C0 [ff 05 c0 08 08 00 01 01]
DEBUG:pyxcp.transport.Base:GET_STATUS
DEBUG:pyxcp.transport.Base:-> [fd]
DEBUG:pyxcp.transport.Base:<- L6 C1 [ff 00 1d 00 00 00]
DEBUG:pyxcp.transport.Base:GET_SEED
DEBUG:pyxcp.transport.Base:-> [f8 00 04]
DEBUG:pyxcp.transport.Base:<- L6 C2 [ff 04 <MY SEED HERE>]
DEBUG:pyxcp.transport.Base:UNLOCK
DEBUG:pyxcp.transport.Base:-> [f7 04 <MY KEY HERE>]
DEBUG:pyxcp.transport.Base:<- L2 C3 [fe 25]
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpResponseError [ERR_ACCESS_LOCKED]
[ERROR (pyxcp.pyxcp.master.errorhandler)]: XcpResponseError [ERR_ACCESS_LOCKED]
ERROR:pyxcp.pyxcp.master.errorhandler:XcpResponseError [ERR_ACCESS_LOCKED]
Try to handle error -- Service: UNLOCK Error-Code: ERR_ACCESS_LOCKED
ERROR:pyXCP:Traceback (most recent call last):
  File "xcp_unlock.py", line 50, in <module>
    result = x.cond_unlock("daq")
  File "/mnt/SD/pyxcp/env/lib/python3.7/site-packages/pyxcp/master/master.py", line 1772, in cond_unlock
    self.unlock(key_length, key)
  File "/mnt/SD/pyxcp/env/lib/python3.7/site-packages/pyxcp/master/errorhandler.py", line 376, in inner
    res = executor(inst, func, arguments)
  File "/mnt/SD/pyxcp/env/lib/python3.7/site-packages/pyxcp/master/errorhandler.py", line 346, in __call__
    preActions, actions, repeater = handler.actions(*getActions(inst.service, self.error_code))
  File "/mnt/SD/pyxcp/env/lib/python3.7/site-packages/pyxcp/master/errorhandler.py", line 258, in actions
    raise UnhandledError("Could not proceed due to unhandled error.")
pyxcp.master.errorhandler.UnhandledError: Could not proceed due to unhandled error.

```

Please if anyone can help I would be very grateful

christoph2, 2023-07-11:
Communication (including .so)  looks fine.
Have you tried
```shell
result = x.cond_unlock() 
```
i.e. unlock every resource, if locked?

neresdu, 2023-07-11:
> Communication (including .so) looks fine. Have you tried
> 
> ```shell
> result = x.cond_unlock() 
> ```
> 
> i.e. unlock every resource, if locked?

Yes i have tried separately with each resource and also with x.cond_unlock() to unlock all one by one but so far i have not been successful. Same error in both cases

danielhrisca, 2023-07-11:
You should use Wireshark to sniff the CANape traffic. Maybe there is a difference

neresdu, 2023-07-11:
> You should use Wireshark to sniff the CANape traffic. Maybe there is a difference

My setup is composed of an ECU connected to a linux hardware where I run the python pyXCP script to collect and send XCP data, in another terminal I follow the sending of information by xcpdump.
I'm not using CANape 

I need to a hw like CANape generating CAN messages to XCP unlock works ?

christoph2, 2023-07-11:
The DLL interfacing was somewhat fooled up -- maybe fixed now.

neresdu, 2023-07-14:
I've solved the problem flashing the correct software for my ECU, it wokrs, the problem was wrong sw for my seed and key file

closed 2023-07-14 by neresdu