christoph2/pyxcp #140: REINIT_DAQ

mortenfc, 2023-08-10:
Hi!

I get a sequence error during alloc_odt of my 2nd daq list.
In the standard it says to REINIT_DAQ
[image]

I can see in the source code this error action is not implemented 
[image]

How do I reinit daq? I can't find any information about it anywhere

christoph2, 2023-08-10:
The only thing that comes to mind, would be FREE_DAQ, but this usually (like in my own Blueparrot XCP project) resets/restarts an internal state-machine, so one has to run FREE_DAQ - ALLOC_DAQ
again.  ERR_SEQUENCE on ALLOC_ODT almost always means that this initialization sequence is missing. Because nobody knows how many DAQ lists the user wants to allocate if ALLOC_DAQ is missing, the only reasonable action is program termination -- OK, I should add an explanatory message.
Your first ALLOC_ODT works? Really strange...

mortenfc, 2023-08-10:
I was doing 

FREE_DAQ
ALLOC_DAQ(2)
ALLOC_ODT
ALLOC_ODT_ENTRY
ALLOC_ODT -> Sequence error

I just saw in the standard you have to call all ALLOC_ODT before ALLOC_ODT_ENTRY :D

christoph2, 2023-08-10:
Yes, to put it into another words:
There's no way back to higher level functions.

s. Blueparrot state-machine:
https://github.com/christoph2/cxcp/blob/master/src/xcp_daq.c#114

mortenfc, 2023-08-10:
This may be out of topic, but what command PID do I use to send a STIM package to a STIM daq list?
There is no command for CommandCategory.STIM

christoph2, 2023-08-11:
STIM functionality isn't implemented yet; due to strict timing requirements in most cases, a C++ extension is needed.
Any ideas how a STIM API could look like, are highly welcome!

mortenfc, 2023-08-11:
Creating a user-friendly and flexible API is probably tricky, since there are many ways to do stimulation, and different setups for different hardware. 
But one could maybe start with

createStimDaqList(dict_of_measurement_var_sizes_and_addresses, dict_of_stimulation_var_sizes_and_addresses)
-> stim_daq_number = 0, meas_daq_number = 1, free_daq -> alloc_daq(2) -> alloc_odt (assuming less than 255 variables) -> alloc_odt -> alloc_odt_entry -> alloc_odt_entry -> set_daq_list_mode(mode=0b00000010) (stim) -> set_daq_list_mode(mode=0b00000000) (meas)

beginStim -> start_stop_synch(mode=0b01)

sendStimDto -> fill header and identification field based on configuration in get_daq_processor_info and the standard, then add payload data of stimulation variables concatenated (this doesn't seem to work for me, I'm not sure what I'm doing wrong)

receiveDaqMeas(slave_cycle_period) -> pop_dto_queue with 1 cycle slave timeout, and parse based on types and order configured in createStimDaqList

christoph2, 2023-08-11:
OK, looks like a good starting point.

Regarding timing, I'll start a separate GH project, including a MIDI arpeggiator!?
There's really a close technical relation between DAQ/STIM and MIDI applications when it comes to timing:
- It makes no difference if a DAQ list is send every 20ms or a drum-machine runs @125BPM, but don't expect to measure exactly 20ms on a non-RTOS... 
The challenge is now to extract the exact period value -- α/β- or even Kalman-filtering, we'll see...
- Playing notes of a chord over and over again at exact points in time corresponds roughly to STIM.

christoph2, 2026-03-01:
**Issue Resolved - REINIT_DAQ Implemented ✅**

This issue has been resolved since **v0.26.3 (2026-02-14)**. The REINIT_DAQ pre-action is now fully implemented and documented.

**Your Original Problem:**
You encountered `ERR_SEQUENCE` during the second `ALLOC_ODT` call because the XCP state machine requires a specific sequence:

**Correct Sequence:**
``
FREE_DAQ
ALLOC_DAQ(2)          # Allocate 2 DAQ lists
ALLOC_ODT(0, n)       # Allocate ODTs for list 0
ALLOC_ODT(1, m)       # Allocate ODTs for list 1
ALLOC_ODT_ENTRY(...)  # Now allocate entries for all ODTs
ALLOC_ODT_ENTRY(...)
...
``

**What Was Wrong (your case):**
``
FREE_DAQ
ALLOC_DAQ(2)
ALLOC_ODT(0, n)
ALLOC_ODT_ENTRY(...) ← Too early! Must finish all ALLOC_ODT first
ALLOC_ODT(1, m)      ← ERR_SEQUENCE - state machine moved to ODT_ENTRY phase
``

**REINIT_DAQ Now Handles This:**
- If you accidentally trigger `ERR_SEQUENCE` or `ERR_MEMORY_OVERFLOW` on `ALLOC_ODT`, the error handler automatically calls `freeDaq()`
- The DAQ configuration is reset to a clean state
- You can retry the allocation sequence

**Implementation:** `pyxcp/master/errorhandler.py` lines 452-456

**For v0.28.0+ Users:**
The NotImplementedError is gone. The library now handles DAQ re-initialization gracefully.

**STIM API:**
Your question about STIM (comment #1673700520) is still an open feature request. STIM functionality requires C++ extensions for timing-critical operations and is not yet implemented. @christoph2's last comment (Aug 2023) indicated it's planned but needs careful API design.

**Closing as the original REINIT_DAQ issue is resolved.** If you need STIM functionality, please open a separate feature request with your use case details.

christoph2, 2026-03-01:
Closing - REINIT_DAQ implemented in v0.26.3, sequence error documented