christoph2/pyxcp #150: XCP-Download command problem

mladjaa, 2023-11-29:
Hello, I am using XCP protocol for communication with ECU in Python.  When I want to write value inside of ECU, I use SET_MTA command, got message from slave "FF" as success, then I use Download command to write value in ECU and always slave respond "FE 23". I found that in error list "Memory location is protected to write". When I try to send same command for same memory location in CANape software, there are no errors (return FF as good and writes value on wished address).
Do you know what is the possible problem for this and how to solve it?

christoph2, 2023-11-30:
Without context information, I would guess a side-effect of an UNLOCK operation?
Could you post the pyXCP code sequence your are using and most important, the CANape trace, esp. during setup?

mladjaa, 2023-12-06:
Here is pictures of Python code and CANape trace window. Please give me your point if view.

[image]

[image]

danielhrisca, 2023-12-06:
You should also capture the CANape connection part

danielhrisca, 2023-12-07:
The excel trace does not contain the TX messages were the XCP commands are sent. Please provide a complete trace recorded when CANape is doing the connect and download

christoph2, 2023-12-07:
Honestly, I don't know why you are completly bypassing pyXCP...
Are you sure this is a CANape trace?
One would expect CANape issuing **GET_STATUS**, **GET_COMM_MODE_INFO**, **GET_ID** commands after **CONNECT**.

mladjaa, 2023-12-08:
Sorry for misunderstanding, trace file which I sent is trace according to Python Code. Now, I am sending trace files from CANape. Commands are in Trace2 file and marked yellow.

[image]

[image]

danielhrisca, 2023-12-08:
[image]
Trace 1 starts after CANape has established the XCP connection

mladjaa, 2023-12-08:
Yes, but this trace is from beginning, I don't know why skiped CONNECT command.

danielhrisca, 2023-12-08:
it is possible that CANape is connecting with mode != 0
[image]

mladjaa, 2023-12-08:
Let's focus please on Python code and DOWNLOAD command problem with error "FE 23"-memory location is protected to write. I found in XCP documentation that parameter which you want to change, first of all, needs to be moved from Flash to RAM, but I am not sure how to do this. 
Have you any idea?

christoph2, 2023-12-13:
The movement is usually done with `SET_CAL_PAGE`. In the first step you should consult your A2L (if available),
`MOD_PAR`/`MEMORY_SEGMENT`s, maybe your address is pageable to RAM.
Optional commands `GET_PAG_PROCESSOR_INFO`, `GET_SEGMENT_INFO`, and `GET_PAGE_INFO` may also helpful.

mladjaa, 2023-12-14:
I resolved problem with command "SET_CAL_PAGE". 
Thank you for your support!