christoph2/pyxcp #155: ERROR:pyxcp.errorhandler:XcpTimeoutError [Response timed out [block_receive].]

MYAN65536, 2023-12-20:
Hi!
When I run the program in the picture, this error is returned, and this error is 'ERROR:pyxcp.errorhandler:XcpTimeoutError [Response timed out [block_receive].] '. 
When I debug the program, I see that the length of self.resQueue is 0.
[image]
[image]

MYAN65536, 2023-12-21:
I've solved the problem, thank you for contributing.

xhbaa, 2024-01-11:
How did you solve it?

MYAN1998, 2024-01-11:
Problem with slave code.

xhbaa, 2024-01-12:
Can you show me your code?

MYAN1998, 2024-01-12:
Sorry.It is the company code.

suannaisdk, 2024-01-19:
@MYAN65536 You can tell us the reason or idea for the fix without showing your code.

I also encountered the same problem, But I tried to modify "CAN_ID_MASTER" and "CAN_ID_SLAVE" in the configuration. Maybe the response time out is caused by you setting the wrong master and slave IDs.

MYAN1998, 2024-01-24:
> @MYAN65536 You can tell us the reason or idea for the fix without showing your code.
> 
> I also encountered the same problem, But I tried to modify "CAN_ID_MASTER" and "CAN_ID_SLAVE" in the configuration. Maybe the response time out is caused by you setting the wrong master and slave IDs.

 就是调试的时候发现的问题，我可以给你看我的配置文件
[image]