christoph2/pyxcp #158: CAN-FD transport MAX_DLC_REQUIRED

danielhrisca, 2024-01-11:
I think that the way `MAX_DLC_REQUIRED` is handled for CAN-FD should be revised. 

In the current implementation this is automatically set to True for CAN-FD (see https://github.com/christoph2/pyxcp/blob/60a38cbea0b1335d205d1655a4c594fcc424557b/pyxcp/transport/can.py#L316)

`SRS_Can_01160` does not state that a maximum frame length  must be used, it only talks about padding the frame up to the next CAN-FD DLC value, which is also correctly implemented in the `pad_frame` function (https://github.com/christoph2/pyxcp/blob/60a38cbea0b1335d205d1655a4c594fcc424557b/pyxcp/transport/can.py#L78)

```
SWS_Can_00502] ⌈ If PduInfo->SduLength does not match possible DLC values 
CanDrv shall use the next higher valid DLC for transmission with initialization of 
unused bytes to the value of the corresponding CanFdPaddingValue (see 
ECUC_Can_00485).⌋ ( SRS_Can_01160)
```

On the other hand the XCP on CAN specification only talks about `MAX_DLC_REQUIRED` which would then result in 8 or 64 bytes frames.

christoph2, 2024-01-11:
Yes, FD-frame padding is only required if SduLength does not exactly matches a discrete DLC value.
The fix will be included in the next big release.

danielhrisca, 2024-01-11:
And only for autosar

Funth0mas, 2024-03-28:
(disclaimer: I don't know XCP-over-CAN specification) The Autosar implementation from EB seem to require minimal DLC=8 on CANfd. It ignores shorter frames. C'est la vie...

christoph2, 2024-03-28:
The FD padding behavior is fixed in the (hopefully soon) coming release.

christoph2, 2026-03-01:
This issue has been resolved. CAN-FD no longer automatically sets `MAX_DLC_REQUIRED=True`.

**Fix Details:**
- **Fixed between:** v0.22.x and v0.28.0
- **Code location:** `pyxcp/transport/can.py` line 477
- **Implementation:** `self.max_dlc_required = self.config.max_dlc_required`

**Current Behavior:**
The `max_dlc_required` parameter is now read from the configuration file, giving users full control over frame padding behavior:

- Set `max_dlc_required = True` → Pad all frames to maximum DLC (8 or 64 bytes)
- Set `max_dlc_required = False` → Use AUTOSAR SWS_Can_00502 padding (pad to next valid DLC)

**Configuration Example:**
``python
from pyxcp.config import XcpConfig

config = XcpConfig(
    transport='CAN',
    can=dict(
        interface='vector',
        fd=True,
        max_dlc_required=False,  # Use smart padding
        # ...
    )
)
``

The `pad_frame()` function correctly implements both padding strategies as discussed.

Closing as resolved.