christoph2/pyxcp #164: About DAQ data

huangbinghao, 2024-08-06:
Hello author, I noticed that you have updated the new branches, Next_Level. Does it support reading DAQ data? If supported, how should run_daq.py run in the example folder?

christoph2, 2024-08-06:
Hello,
I'm glad you're asking!
Yes, there's a really big update coming and a key feature is DAQ handling.
While I did extensive testing with XCPlite, CANApe and my own cXCP and everything looks really good, I can't release without some sort of documentation (for example, the configuration system is also completly new).
On the other hand, I'm also working hard on a big pya2ldb update.
OK, let's see if I'm able to put together enough documentation within this week -- "next_level" is basically in a 
release candidate state!

huangbinghao, 2024-08-06:
Thanks!!!!! 
I am looking forward to the update of this version very much

christoph2, 2026-03-01:
Closing - next_level merged, DAQ fully available in v0.27.0+

closed 2026-03-01 by christoph2