christoph2/pyxcp #167: pyxcp build error with pip in ubuntu

JavierCorado, 2024-08-26:
Attempted to install the latest version of pyxcp (v0.22.1) with pip (pip install pyxcp) and there are issues while building the package:

[image]

Saw this in both ubuntu WSL and ubuntu version 22.04.1 running on: x86_64 x86_64 x86_64 GNU/Linux

christoph2, 2024-08-27:
Hello,

The build-system requires `cmake`, which is missing in your case.

Building pyXCP  on Linux  is a bit tricky, so I did some Docker experiments and came to the following requirements:

```shell
sudo apt install -y pkg-config
sudo apt install -y gcc git libssl-dev python3 python3-pip python3-poetry rustc cargo libffi-dev cmake
````

And yes, there is a pro-forma source distro for `v22.1`  on pypi, but I doubt it is complete, so you should 
start with a cloned/forked repo.

There will be some smaller updates in the next time (including pre-build Linux packages), so make sure 
to install pyXCP in editable mode:

```shell
pip3 install -e .
```

The next thing I observed on Linix:
Even if `poetry` should install `pybind11`, `cmake`  may unable to locate it, in this case, just manually

```python
pip3 install pybind11
```

Hope it helps.

JavierCorado, 2024-08-27:
I was able to Install pyxcp v0.22.1 locally with `pip3 install -e .` I will be taking a look at those new features that v0.22.1 has

Thanks for the help!

closed 2024-08-27 by JavierCorado