christoph2/pyxcp #169: Cannot build pyxcp with python3.12 on ubuntu 24.04

sff0219, 2024-09-17:
I can't build `pyxcp` with python 3.12 on ubuntu 24.04. Got the following error:
````
Building wheels for collected packages: pyxcp, uptime
  Building wheel for pyxcp (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pyxcp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [43 lines of output]
      Platform Linux
      build_ext::build_extension()
       BUILD-TYPE: 'Release'
      ================================================================================
                                     Step #1: Configure
      ================================================================================
      Traceback (most recent call last):
        File "/tmp/pip-req-build-pw9w4san/build_ext.py", line 121, in <module>
          build_extension(False)
        File "/tmp/pip-req-build-pw9w4san/build_ext.py", line 105, in build_extension
          subprocess.run(["cmake", "-S", str(TOP_DIR), *cmake_args], cwd=build_temp, check=True)  # nosec
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3.12/subprocess.py", line 548, in run
          with Popen(*popenargs, **kwargs) as process:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3.12/subprocess.py", line 1026, in __init__
          self._execute_child(args, executable, preexec_fn, close_fds,
        File "/usr/lib/python3.12/subprocess.py", line 1955, in _execute_child
          raise child_exception_type(errno_num, err_msg, err_filename)
      FileNotFoundError: [Errno 2] No such file or directory: 'cmake'
      Traceback (most recent call last):
        File "/home/ubuntu/.local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/ubuntu/.local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/ubuntu/.local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-t68yisgi/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/api.py", line 58, in build_wheel
          return WheelBuilder.make_in(
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-t68yisgi/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 88, in make_in
          wb.build(target_dir=directory)
        File "/tmp/pip-build-env-t68yisgi/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 123, in build
          self._build(zip_file)
        File "/tmp/pip-build-env-t68yisgi/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 172, in _build
          self._run_build_script(self._package.build_script)
        File "/tmp/pip-build-env-t68yisgi/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 262, in _run_build_script
          subprocess.check_call([self.executable.as_posix(), build_script])
        File "/usr/lib/python3.12/subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['/usr/bin/python3.12', 'build_ext.py']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyxcp
  Building wheel for uptime (setup.py) ... done
  Created wheel for uptime: filename=uptime-3.0.1-cp312-cp312-linux_x86_64.whl size=6371 sha256=d8577aebb29d5f4b5d528039014cf2ca99b8986397b3521048a20c502107a430
  Stored in directory: /home/ubuntu/.cache/pip/wheels/57/68/57/3e2c834bd7abbdc4b37335132ab36b78d0cd9f4525200c73ef
Successfully built uptime
Failed to build pyxcp
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (pyxcp)
````

christoph2, 2024-09-17:
The error message means `cmake` is missing.

`sudo apt install cmake` 
should fix this issue.

P.S.: I just fixed a last minute issue -- if you experience problems locating you Python installation, make shure to get the latest pyXCP version.

sff0219, 2024-09-17:
> The error message means `cmake` is missing.
> 
> `sudo apt install cmake` should fix this issue.
> 
> P.S.: I just fixed a last minute issue -- if you experience problems locating you Python installation, make shure to get the latest pyXCP version.

Thanks. I overlooked the error message. However I got a new error while trying to install it. 

````
  Building wheel for pyxcp (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pyxcp (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [36 lines of output]
      Platform Linux
      build_ext::build_extension()
       BUILD-TYPE: 'Release'
      Traceback (most recent call last):
        File "/tmp/pip-req-build-yjgqccz1/build_ext.py", line 140, in <module>
          build_extension(False)
        File "/tmp/pip-req-build-yjgqccz1/build_ext.py", line 96, in build_extension
          py_cfg = get_py_config()
                   ^^^^^^^^^^^^^^^
        File "/tmp/pip-req-build-yjgqccz1/build_ext.py", line 80, in get_py_config
          return dict(exe=sys.executable, include=include, libdir=libdir, library=library)
                                                                  ^^^^^^
      UnboundLocalError: cannot access local variable 'libdir' where it is not associated with a value
      Traceback (most recent call last):
        File "/home/ubuntu/.local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/ubuntu/.local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/ubuntu/.local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-jto2a7hr/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/api.py", line 58, in build_wheel
          return WheelBuilder.make_in(
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-jto2a7hr/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 88, in make_in
          wb.build(target_dir=directory)
        File "/tmp/pip-build-env-jto2a7hr/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 123, in build
          self._build(zip_file)
        File "/tmp/pip-build-env-jto2a7hr/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 172, in _build
          self._run_build_script(self._package.build_script)
        File "/tmp/pip-build-env-jto2a7hr/overlay/local/lib/python3.12/dist-packages/poetry/core/masonry/builders/wheel.py", line 262, in _run_build_script
          subprocess.check_call([self.executable.as_posix(), build_script])
        File "/usr/lib/python3.12/subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['/usr/bin/python3.12', 'build_ext.py']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyxcp
Failed to build pyxcp
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (pyxcp)
````
 Did I miss anything again?

christoph2, 2024-09-17:
OK, there are still problems locating the Python link library.

The unbound variable `libdir` means the search process was unsuccessful.
The latest commit contains a small update, hopefully it works.

sff0219, 2024-09-20:
Hi,

I still couldn't build it. Somehow CMake could not find python library. 

As a solution from my side, I have to use an older but stable version 0.21.10

JavierCorado, 2024-09-20:
@sff0219 
You can try a workaround shared by @christoph2 in another issue: https://github.com/christoph2/pyxcp/issues/167

Installing the dependencies manually and installing the pyxcp package locally did the trick for me.

christoph2, 2024-09-21:
One thing came to mind:

Probably you are running a Python version without development libraries.

You may verify this, if you run first

```shell
find /usr -name "libpython*.so"
```

and then

```shell
 find /usr -name "libpython*.a"
```
the second one is important; an empy response means that the link librarary requirered for extension building is missing.
Look for package `libpython3-dev`, `python3-dev` or similar.