christoph2/pyxcp #175: when used the shortDownload occur error

wangwei357, 2024-10-11:
The test environment:
    config:   "TRANSPORT": "CAN","CAN_DRIVER": "Vector",
the XCP slave was xcpsim.exe where from  vector CANape，
when I try used the shortDownload function like this: 
            data = [0xCA, 0xFE, 0xBA, 0xBE]
            res = xcp_master.shortDownload(0x12345678, 0x55, data)
the error was occur ,
the error info is below: RuntimeError: (c_ubyte_Array_8) IndexError: invalid index
 xl_event.tagData.msg.data = tuple(msg.data)  
it''s seem that the can message out of DLC.
how can i solve it

[image]

christoph2, 2024-10-11:
Well, `SHORT_DOWNLOAD`  is in this case simply not possible -- to quote the XCP spec.:
```
Please note that this command will have no effect (no data bytes can be transferred) if
MAX_CTO = 8 (e.g. XCP on CAN).
```
But there is a `push` method, which transparently handles aspects like maximum payload length and XCP block-mode.

```python
xcp_master.push(address, data)
```

christoph2, 2024-10-11:
Note:
Paramter `address_ext` was missing but is now included in the latest release (0.22.10), i.e.:

```python
def push(self, address: int, address_ext: int, data: bytes, callback=None):
    ...
```

wangwei357, 2024-10-11:
@christoph2  Thanks for your replay, I will check the ' xcp_master.push(address, data) '

closed 2024-10-11 by wangwei357