christoph2/pyxcp #176: My previous logging to file function will be affected and cannot be used. The previous log.info function reported errors

wangwei357, 2024-10-12:
My previous logging to file function will be affected and cannot be used. The previous other  log.info function reported errors

  File "C:\Users\XXXXXX\AppData\Local\anaconda3\Lib\logging\__init__.py", line 1113, in emit
    stream.write(msg + self.terminator)
ValueError: I/O operation on closed file.

wangwei357, 2024-10-12:
It‘s seems that :  myself  logger file handler was force closed when setup the pyxcy application.

christoph2, 2024-10-12:
Yes, pyXCP uses `traitlets` under the hood, which is used by projects like IPython and Jupyter, so the likehood that there is a hidden bug is really small. Maybe there's a similar logging issue on Stackoverflow?
For now you may just swallow the exception:
```python
try:
   # Offending code goes here.
except ValueError as e:
   print(f"Logging error {e!r}")
```

christoph2, 2024-10-12:
The effect of closing can be easily observed...

```shell
python -c "import sys as s;s.stdout.close();print('hello')"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ValueError: I/O operation on closed file.
```

wangwei357, 2024-10-12:
@christoph2  Can I disable pyxcp logger function? Now this feature applies to my use of other programs

christoph2, 2024-10-12:
At the first sight, there are two places which could be commented out:

https://github.com/christoph2/pyxcp/blob/0bcf71455d24d29b3261560ec32b7528477f6110/pyxcp/config/__init__.py#L941

https://github.com/christoph2/pyxcp/blob/0bcf71455d24d29b3261560ec32b7528477f6110/pyxcp/config/__init__.py#L956

If it helps, I'll add a configuration option.

christoph2, 2024-10-12:
Also interesting would be something like

```Python
logger = logging.getLogger("my-own-logger")
```
as a starting point for your logging stuff.

christoph2, 2024-10-12:
I've added a [small test](https://github.com/christoph2/pyxcp/commit/fedec4ba6418c52686eeb73ca9186481d8146bf5) for a already configured root handler, if so use it.
Hope it helps.

wangwei357, 2024-10-14:
@christoph2  Thansk for your support !  I had try your method, It can solve the problem of error, but the function of affecting the file output to the file is no longer available. I use debug to locate the problem statement from this location.   
https://github.com/christoph2/pyxcp/blob/41d6fd9cc96d4c5fd830fedd30886ad818468c28/pyxcp/config/__init__.py#L970C3-L970C62

this  self.read_configuration_file(file_name,False)  in  _read_configuration(self, file_name: str, emit_warning: bool = True) -> None
 made my other logger file handler error

wangwei357, 2024-10-14:
Fortunately, I solved the problem by adjusting the search. However, this logger is redirected to the root logger, I do not know if there will be other effects
```python
            # self._setup_logger()
            if logging.getLogger().hasHandlers():
                self.log = logging.getLogger()
            else:
                self._setup_logger()
```

christoph2, 2024-10-14:
```python
if logging.getLogger().hasHandlers():
    self.log = logging.getLogger()
else:
    self._setup_logger()
self._read_configuration(self.config_file)
```
vs.

```python
self._read_configuration(self.config_file)
if logging.getLogger().hasHandlers():
    self.log = logging.getLogger()
else:
    self._setup_logger()

```
Really makes a difference?

The other thing that came to mind: I could add a `use_existing_logger` configuration option.

wangwei357, 2024-10-14:
Yea，It's make difference