christoph2/pyxcp #184: Use PyXCP to calibrate and measure an ECU

PCT-VF, 2024-11-26:
I have an A2L file that contains a list of variables and their corresponding 'ECU Addresses.' For example:
[https://github.com/christoph2/pyA2L/blob/master/examples/example-a2l-file.a2l](https://github.com/christoph2/pyxcp/issues/url).
Now, I need to use CAN XCP to read/write the values of these variables based on their addresses. I have reviewed the examples of PyXCP, but they do not work. Could you help provide an example relevant to my requirements? Thank you very much.

JavierCorado, 2024-11-26:
Here is a general guideline for reading and writing once you successfully stablished connecting with your slave:

For reading you have two options, polling and DAQ. If you want to read data through polling you can user the shortUpload and upload methods, the size of the XCP packet determines which one you should use (you can calculate the size of the XCP  packet with the data type of the signal inside of the A2L).

For writing, just like in poll based reading you can use shortDownload and download methods (again you have to calculate the size of the XCP packet just like I mentioned above). Just make sure to pack the data with the corresponding length and data type (you can find this information inside the variable definition in the A2L file).

Note: Be aware if the signal is an array of values or a single value, this affects the size calculation.

shirayr, 2024-11-26:
@JavierCorado Hi thank trying to use this solution for my case(https://github.com/christoph2/pyxcp/discussions/183)

but iam failing with initiate the argument parser:
ap = ArgumentParser()
with ap.run() as x:
    x.connect()
    
  iam requested to create the conf_file: pyxcp_conf.py
  
  what should i put there if these are my configurations:
  i use ethernet, UDP communication, IPV6 host
receive on:
Debug PC
IPv4: 192.168.X.X
IPv6: XXXXXXXXXXXXXXXX
UDP Port: 5555
VLANID: 2
MAC Address: XXXXXXXxX

send to ECU:
PH XCP Slave (KVS):
IPv6: XXXXXXXXXX
UDP Port: XXXX

christoph2, 2024-11-27:
Well, `pyXCP` s basically about XCP and `pya2ldb` is about A2L...
The interaction of these and other projects will be the task of [asamint](https://github.com/christoph2/asamint).

I've created a small example to show the basic procedure of running a DAQ list measurement (Tested with [XCPlite](https://github.com/vectorgrp/XCPlite)).

Note: pya2ldb uses SQLAlchemy under the hood, so arbitrary complex queries are possible, but for the sake of simplicity the example code selects all `MEASUREMENT`s.

Tip:  There is a small command-line utility [xcp_fetch_a2l](https://github.com/christoph2/pyxcp/blob/master/pyxcp/scripts/xcp_fetch_a2l.py) that can be used to fetch the correspondending A2L file.

```python

import time

from pyxcp.cmdline import ArgumentParser
from pyxcp.daq_stim import DaqList, DaqRecorder

from pya2l import DB
from pya2l import model

A2L_FILE = "C_Demo" # No extension -- .a2l is assumed.

db = DB()
session = db.open_create(A2L_FILE)  # Import A2L or open existing .a2ldb

ASAM_TYPE_MAP = {
    "UBYTE": "U8",
    "SBYTE": "I8",
    "UWORD": "U16",
    "SWORD": "I16",
    "ULONG": "U32",
    "SLONG": "I32",
    "A_UINT64": "U64",
    "A_INT64": "I64",
    "FLOAT16_IEEE": "F16",
    "FLOAT32_IEEE": "F32",
    "FLOAT64_IEEE": "F64",
}        
    
measurements = []
for m in session.query(model.Measurement).order_by(model.Measurement.name).all():
    name = m.name
    address = m.ecu_address.address
    address_ext = m.ecu_address_extension.extension if m.ecu_address_extension else 0
    asam_type = m.datatype
    data_type = ASAM_TYPE_MAP.get(asam_type)

    # You may check here if the measurement should be used.
    measurements.append((name, address, address_ext, data_type, ))
    print(f"MEASUREMENT:  {name:<20s} address: 0x{address:08x}:{address_ext}  -- type: {asam_type!r}")

ap = ArgumentParser(description="DAQ test")

EVENT_CHANNEL = 0

daq_lists = [
    DaqList(
        "demo_measurement",
        EVENT_CHANNEL,
        False, # List is DAQ not STIM.
        True,  # Use DAQ timestamps if available.
        measurements,
    )
]

daq_parser = DaqRecorder(daq_lists, "demo_measurement", 1)

with ap.run(policy=daq_parser) as x:
    x.connect()
    if x.slaveProperties.optionalCommMode:
        x.getCommModeInfo()

    x.cond_unlock("DAQ")  # DAQ resource is locked in many cases.

    print("setup DAQ lists.")
    daq_parser.setup()
    print("start DAQ lists.")
    daq_parser.start()

    time.sleep(5 * 60)  # Run for 5 minutes.

    print("Stop DAQ....")
    daq_parser.stop()
    print("finalize DAQ lists.\n")
    x.disconnect()
```

christoph2, 2024-11-27:
Regarding basic setup:

First create a configuration file:
```shell
xcp-profile create -o pyxcp_conf.py
```
Then edit `pyxcp_conf.py`, the relevant lines in case of XCPonEth are:

```python
c.Transport.layer="ETH"

c.Transport.Eth.port=5555
c.Transport.Eth.protocol="UDP"  # or TCP

c.Transport.Eth.host="localhost"  # name or IP of XCP slave

# c.General.seed_n_key_dll="SeedNKeyXcp.dll"   # optional name of Seed-and-Key .dll
```
A good starting point is the script [xcp_info](https://github.com/christoph2/pyxcp/blob/master/pyxcp/scripts/xcp_info.py), which gathers some useful information (If a Seed-and-Key .dll is required, you'll notice here...)

christoph2, 2024-11-27:
I forget to mention:
If the measurement was successful, one can use  ex_arrow.py, ex_csv.py, ex_excel.py, ex_mdf.py, ex_sqlite.py to convert from the measurement file (`demo_measurement.xmraw` in this case) to common file formats.

The necessary dependencies like `pyarrow`, `asammdf`, ... are not installed by pyXCP.

shirayr, 2024-11-28:
Hi thanks !!
2 questions:
1
my pyxcp_conf looks line:
c = get_config()  # noqa
c.Transport.layer="ETH"

c.Transport.Eth.protocol="UDP"  # or TCP
c.Transport.Eth.ipv6 = True
c.Transport.Eth.bind_to_address="fd53:........masterIPV6"
c.Transport.Eth.bind_to_port=5556
## Hostname or IP address of XCP slave.
c.Transport.Eth.host = "FD53:......slave ip"
c.Transport.Eth.port=5556 **- his is the master port? where should i insert the slave port??**

2. trying to run pyxcp/scripts/xcp_info.py , stuck on:
\pyxcp\master\errorhandler.py
executor = Executor()
res = executor(inst, func, arguments)
why?

PCT-VF, 2024-12-04:
I use this code: 
```python
import time
from pyxcp.cmdline import ArgumentParser
from pyxcp.daq_stim import DaqList, DaqRecorder
import logging

logging.basicConfig(level=logging.DEBUG)

# Define measurements directly
measurements = [
    ("COM_WindowPositionFL", 0x3434CA50, 0, "U8"),
    ("COM_WindowLockDSBBtnPress", 0x3434CA51, 0, "U8"),
    ("COM_WindowFRThermalProtect", 0x3434CA52, 0, "U8"),
    ("COM_WindowFRSuppressionWarn", 0x3434CA53, 0, "U8"),
]

logging.info("DAQ Measurements: %s", measurements)

daq_lists = [
    DaqList(
        "demo_measurement",
        0,    # EVENT_CHANNEL
        False,  # List is DAQ not STIM.
        True,   # Use DAQ timestamps if available.
        measurements,
    )
]

logging.info("Init DAQ Argument Parser")
ap = ArgumentParser(description="DAQ test")

daq_parser = DaqRecorder(daq_lists, "demo_measurement", 1)
logging.info("Initialized DAQ Recorder")

try:
    with ap.run(policy=daq_parser) as x:
        # logging.info("Connecting to ECU...")
        x.connect()
        # logging.info("Connected to ECU via XCP on CAN")
        if x.slaveProperties.optionalCommMode:
            x.getCommModeInfo()

        x.cond_unlock("DAQ")  # Unlock DAQ resource if locked.

        # logging.info("Setting up DAQ lists...")
        try:
            daq_parser.setup()
        except Exception as e:
            logging.error(f"Failed to setup DAQ lists: {e}")
        
        # logging.info("Starting DAQ...")
        daq_parser.start()

        time.sleep(60)  # Run for 60 seconds.

        # logging.info("Stopping DAQ...")
        daq_parser.stop()
        # logging.info("Finalizing DAQ lists.")
        x.disconnect()
except Exception as e:
    logging.error(f"An error occurred: {e}")
```
And the result is:
```command
PS C:\Tools\ACNI\DAL\utlis> python .\delete_A2L.py
INFO:root:DAQ Measurements: [('COM_WindowPositionFL', 875874896, 0, 'U8'), ('COM_WindowLockDSBBtnPress', 875874897, 0, 'U8'), ('COM_WindowFRThermalProtect', 875874898, 0, 'U8'), ('COM_WindowFRSuppressionWarn', 875874899, 0, 'U8')]
INFO:root:Init DAQ Argument Parser
DEBUG:root:Looking for pyxcp_conf in C:\Tools\ACNI\DAL\utlis
DEBUG:root:Loaded config file: C:\Tools\ACNI\DAL\utlis\pyxcp_conf.py
INFO:root:Initialized DAQ Recorder
INFO:can.interfaces.vector.canlib:Found 12 channels
INFO:can.interfaces.vector.canlib:Channel index 1: VN8917 Channel 1
INFO:can.interfaces.vector.canlib:Channel index 2: VN8917 Channel 2
INFO:can.interfaces.vector.canlib:Channel index 5: VN8917 Channel 5
INFO:can.interfaces.vector.canlib:Channel index 6: VN8917 Channel 6
INFO:can.interfaces.vector.canlib:Channel index 7: VN8917 Channel 7
INFO:can.interfaces.vector.canlib:Channel index 8: VN8917 Channel 8
INFO:can.interfaces.vector.canlib:Channel index 10: Virtual Channel 1
INFO:can.interfaces.vector.canlib:Channel index 11: Virtual Channel 2
DEBUG:can.interface.detect_available_configs:interface "vector" detected 8 available configurations      
INFO:root:XCPonCAN - Interface-Type: 'vector' Parameters: [('channel', 0), ('fd', False), ('bitrate', 500000), ('receive_own_messages', False), ('app_name', 'python-can')]
INFO:root:XCPonCAN - Master-ID (Tx): 0x000007B1S -- Slave-ID (Rx): 0x00000782S
DEBUG:can.interfaces.vector.canlib:Channel index 0 found
DEBUG:can.interfaces.vector.canlib:Open Port: PortHandle: 0, ChannelMask: 0x2, PermissionMask: 0x2
INFO:can.interfaces.vector.canlib:xlCanSetChannelBitrate: baudr.=500000
INFO:root:XCPonCAN - Using Interface: 'Application python-can: CAN 1'
INFO:root:XCPonCAN - Filters used: [{'can_id': 1922, 'can_mask': 2047, 'extended': False}]
INFO:root:XCPonCAN - State: BusState.ACTIVE
DEBUG:root:CONNECT
DEBUG:root:-> [ff 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C0 [ff 15 40 08 08 00 01 01]
DEBUG:root:GET_STATUS
DEBUG:root:-> [fd 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C1 [ff 00 00 00 00 00 ff ff]
DEBUG:root:<- L8 C1 [ff 00 00 00 00 00 ff ff]

DEBUG:root:GET_DAQ_PROCESSOR_INFO
DEBUG:root:GET_DAQ_PROCESSOR_INFO
DEBUG:root:-> [da 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C2 [ff 89 06 00 05 00 00 00]
DEBUG:root:GET_DAQ_RESOLUTION_INFO
DEBUG:root:-> [d9 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C3 [ff 01 07 01 00 64 01 00]
DEBUG:root:<- L8 C3 [ff 01 07 01 00 64 01 00]
DEBUG:root:GET_DAQ_EVENT_INFO
DEBUG:root:-> [d7 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C4 [ff 44 01 14 0a 06 09 ff]
DEBUG:root:UPLOAD
DEBUG:root:-> [f5 14]
DEBUG:root:SYNCH
DEBUG:root:-> [fc 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C5 [fe 00 ff ff ff ff ff ff]
ERROR:root:Failed to setup DAQ lists: Failed to retrieve DAQ Info: required argument is not an integer   
DEBUG:root:START_STOP_SYNCH
DEBUG:root:-> [dd 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C6 [ff ff ff ff ff ff ff ff]
DEBUG:root:START_STOP_SYNCH
DEBUG:root:-> [dd 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C7 [ff ff ff ff ff ff ff ff]
DEBUG:root:DISCONNECT
DEBUG:root:-> [fe 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C8 [ff ff ff ff ff ff ff ff]
```
After debug, I found the error " Failed to retrieve DAQ Info: required argument is not an integer" occurs in the following function: 
```python 
daq_parser.setup() 
     self.daq_info = self.xcp_master.getDaqInfo() #daq_stim __init__.py
         name = self.fetch(eci.eventChannelNameLength) # master master.py
              data = self.upload(remaining) # master master.py
                    response = self.transport.request(types.Command.UPLOAD, length) # master master.py
````
Currently, I am unable to resolve this issue. Can you help me?

christoph2, 2024-12-04:
Is the `SYNCH` issue repeateable?
Could you disable the try/except block, an execption trace would be really helpful.
You could also try to run `xcp-info`, maybe the outcome is different...

PCT-VF, 2024-12-05:
Thanks for your hint. After I removed `try - except` block, the result is:
```command
Adress: None
ERROR:root:Traceback (most recent call last):
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\daq_stim\__init__.py", line 42, in setup
    self.daq_info = self.xcp_master.getDaqInfo() #ERROR HERE
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.py", line 1732, in getDaqInfo
    name = self.fetch(eci.eventChannelNameLength) # ERROR HERE
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.py  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.py", line 594, in fetch
    data = self.upload(remaining) # ERROR HERE
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 431, in inner
", line 594, in fetch
    data = self.upload(remaining) # ERROR HERE
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 431, in inner
    res = executor(inst, func, arguments)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 431, in inner
    res = executor(inst, func, arguments)
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 431, in inner
    res = executor(inst, func, arguments)
ler.py", line 431, in inner
    res = executor(inst, func, arguments)
    res = executor(inst, func, arguments)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 363, in __call__
ler.py", line 363, in __call__
    res = handler.execute()
          ^^^^^^^^^^^^^^^^^
          ^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 230, in execute
ler.py", line 230, in execute
    return self.func(self.instance, *self.arguments.args, **self.arguments.kwargs)
    return self.func(self.instance, *self.arguments.args, **self.arguments.kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.py", line 415, in setMta
    addr = self.DWORD_pack(address)
           ^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: required argument is not an integer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Tools\ACNI\DAL\utlis\delete_A2L.py", line 49, in <module>
    daq_parser.setup()
  File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\daq_stim\__init__.py", line 44, in setup
    raise RuntimeError(f"Failed to retrieve DAQ Info: {e}")
RuntimeError: Failed to retrieve DAQ Info: required argument is not an integer

ERROR:root:An error occurred: Failed to retrieve DAQ Info: required argument is not an integer
```
I saw that variable `address` is None
How can I fix it?

PCT-VF, 2024-12-05:
My final proposal is to see the values of all measurements and characteristics on the ECU, which are defined in the A2L file

christoph2, 2024-12-05:
Hard to diagnose with zero information...

- There is a timeout at the UPLOAD command followed by a SYNCH command, this seems reproduceable; you haven't told me.
- Have you ever tried to run  without `loglevel=DEBUG`? 
   logging  has currently an adverse effect on performance, don't activate for no reason!
   If the error still persists, probably something is wrong with your XCP slave.
- I don't know what is the output of `xcp-info`.
- Looks like a plain Python exception trace -- did you for any reason disable `rich.traceback`?

PCT-VF, 2024-12-05:
When I try to run `xcp-info.py` the result is similar:
```python
PS C:\Tools\ACNI\DAL\utlis> python .\xcp-info.py
2024-12-05 11:05:55 INFO     XCPonCAN - Interface-Type: 'vector' Parameters: [('channel', 0), ('fd', False), ('bitrate', 500000), ('receive_own_messages',     can.py:333
                             False), ('app_name', 'python-can')]
                    INFO     XCPonCAN - Master-ID (Tx): 0x000007B1S -- Slave-ID (Rx): 0x00000782S                                                              can.py:334
2024-12-05 11:05:56 INFO     XCPonCAN - Using Interface: 'Application python-can: CAN 1'                                                                       can.py:256
                    INFO     XCPonCAN - Filters used: [{'can_id': 1922, 'can_mask': 2047, 'extended': False}]                                                  can.py:257
                    INFO     XCPonCAN - State: BusState.ACTIVE                                                                                                 can.py:258

Slave Properties:
=================
{'addressGranularity': EnumIntegerString.new(0, 'BYTE'),
 'byteOrder': EnumIntegerString.new(0, 'INTEL'),
 'bytesPerElement': 1,
 'maxCto': 8,
 'maxDto': 8,
 'maxWriteDaqMultipleElements': 0,
 'optionalCommMode': False,
 'pgmProcessor': {},
 'protocolLayerVersion': 1,
 'slaveBlockMode': True,
 'supportsCalpag': True,
 'supportsDaq': True,
 'supportsPgm': True,
 'supportsStim': False,
 'transportLayerVersion': 1,
 'transport_layer': 'CAN'}

Implemented IDs:
================

Protection Status
=================
    dbg   : False
    pgm   : False
    stim  : False
    daq   : False
    calpag: False

DAQ Info:
=========

Adress: None
2024-12-05 11:05:58 ERROR    Traceback (most recent call last):                                                                                             master.py:130
                               File "C:\Tools\ACNI\DAL\utlis\xcp-info.py", line 37, in main
                                 daq_info = x.getDaqInfo()
                                            ^^^^^^^^^^^^^^
                               File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.py", line 1732, in      
                             getDaqInfo
                                 name = self.fetch(eci.eventChannelNameLength) # ERROR HERE
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                               File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.py", line 594, in fetch 
                                 data = self.upload(remaining) # ERROR HERE
                                        ^^^^^^^^^^^^^^^^^^^^^^
                               File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 431, in 
                             inner
                                 res = executor(inst, func, arguments)
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                               File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 363, in 
                             __call__
                                 res = handler.execute()
                                       ^^^^^^^^^^^^^^^^^
                               File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\errorhandler.py", line 230, in 
                             execute
                                 return self.func(self.instance, *self.arguments.args, **self.arguments.kwargs)
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                               File "C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.py", line 415, in       
                             setMta
                                 addr = self.DWORD_pack(address)
[... 69 lines cut ...]
│   233 │   │   """Preprocess errorhandling pre-actions and actions."""                            │
│                                                                                                  │
│ ╭──────────────────────────────── locals ─────────────────────────────────╮                      │
│ │ self = <pyxcp.master.errorhandler.Handler object at 0x000001DC5BABE9F0> │                      │
│ ╰─────────────────────────────────────────────────────────────────────────╯                      │
│                                                                                                  │
│ C:\Users\trangpc\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyxcp\master\master.p │
│ y:415 in setMta                                                                                  │
│                                                                                                  │
│    412 │   │   """                                                                               │
│    413 │   │   self.mta = types.MtaType(address, addressExt)  # Keep track of MTA (needed for e  │
│    414 │   │   print(f"Adress: {address}")                                                       │
│ ❱  415 │   │   addr = self.DWORD_pack(address)                                                   │
[... cut ...]
│ │    address = None                                                      │                       │
│ │ addressExt = None                                                      │                       │
│ │       self = <pyxcp.master.master.Master object at 0x000001DC5B98B1A0> │                       │
│ ╰────────────────────────────────────────────────────────────────────────╯                       │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
error: required argument is not an integer
```

christoph2, 2024-12-05:
Maybe really an configuration issues ? 
An empty response from `GET_ID` /  *Implemented IDs*  is rather uncommon...

Also, to preclude a systematic error of the `UPLOAD` command you should try to read a known-good address (from you A2L),
something like this:
```python
LEN = 4
ADDR = 4711
x.setMta(ADDR)
data = x.fetch(LEN)
```

christoph2, 2024-12-05:
You should also give [XCPlite](https://github.com/vectorgrp/xcplite) a chance, to have a working reference system (I'll include a `Dockerfile` in the next release, that simply works...)

PCT-VF, 2024-12-10:
Currently, I have three issues that I need your support with.

1. `daq_parser.start()` issue

I resolved the previous issue, but now I’m facing a new problem: I cannot stop the DAQ because ```start_stop_synch``` is not functioning.My program often stops immediately after the `daq_parser.start()` command, even though it still sends the message DD 01. It doesn’t display any error for this issue, whether DEBUG mode is enabled or not.
```python
DEBUG:root:FREE_DAQ
DEBUG:root:-> [d6 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C8 [ff ff ff ff ff ff ff ff]
dq: b'\x01\x00\x00\x00\x00\x00'
DEBUG:root:ALLOC_DAQ
DEBUG:root:-> [d5 00 01 00 00 00 00 00]
DEBUG:root:<- L8 C9 [ff ff ff ff ff ff ff ff]
DEBUG:root:ALLOC_ODT
DEBUG:root:-> [d4 00 00 00 01 00 00 00]
DEBUG:root:<- L8 C10 [ff ff ff ff ff ff ff ff]
DEBUG:root:ALLOC_ODT_ENTRY
DEBUG:root:-> [d3 00 00 00 00 01 00 00]
DEBUG:root:<- L8 C11 [ff ff ff ff ff ff ff ff]
DEBUG:root:SET_DAQ_PTR
DEBUG:root:-> [e2 00 00 00 00 00 00 00]
DEBUG:root:<- L8 C12 [ff ff ff ff ff ff ff ff]
DEBUG:root:WRITE_DAQ
DEBUG:root:-> [e1 ff 01 00 99 cb 34 34]
DEBUG:root:<- L8 C13 [ff ff ff ff ff ff ff ff]
DEBUG:root:SET_DAQ_LIST_MODE
DEBUG:root:-> [e0 00 00 00 00 00 01 06]
DEBUG:root:<- L8 C14 [ff ff ff ff ff ff ff ff]
DEBUG:root:START_STOP_DAQ_LIST
DEBUG:root:-> [de 02 00 00 00 00 00 00]
DEBUG:root:<- L8 C15 [ff 00 ff ff ff ff ff ff]
Send request startStopSynch
DEBUG:root:START_STOP_SYNCH
DEBUG:root:-> [dd 01 00 00 00 00 00 00]
DEBUG:root:<- L8 C16 [ff ff ff ff ff ff ff ff]
DEBUG:root:<- L8 C17 ODT_Data[0:8] [00 02 ff ff ff ff ff ff]
PS C:\Tools\ACNI\DAL\utlis> 
```
Here are the messages on the CAN network for your easier reference:
```python
[MASTER]: Timestamp: 1733813801.753901    ID:      7b1    S Rx                DL:  8    ff 00 00 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.754621    ID:      782    S Rx                DL:  8    ff 15 40 08 08 00 01 01     Channel: 0
[MASTER]: Timestamp: 1733813801.756964    ID:      7b1    S Rx                DL:  8    da 00 00 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.764632    ID:      782    S Rx                DL:  8    ff 89 06 00 05 00 00 00     Channel: 0
[MASTER]: Timestamp: 1733813801.766844    ID:      7b1    S Rx                DL:  8    d9 00 00 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.774618    ID:      782    S Rx                DL:  8    ff 01 07 01 00 64 01 00     Channel: 0
[MASTER]: Timestamp: 1733813801.776600    ID:      7b1    S Rx                DL:  8    d7 00 00 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.784604    ID:      782    S Rx                DL:  8    ff 44 01 14 0a 06 09 ff     Channel: 0
[MASTER]: Timestamp: 1733813801.786455    ID:      7b1    S Rx                DL:  8    d7 00 01 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.794615    ID:      782    S Rx                DL:  8    ff 44 01 14 02 07 08 ff     Channel: 0
[MASTER]: Timestamp: 1733813801.796491    ID:      7b1    S Rx                DL:  8    d7 00 02 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.804625    ID:      782    S Rx                DL:  8    ff 44 01 14 05 07 07 ff     Channel: 0
[MASTER]: Timestamp: 1733813801.806649    ID:      7b1    S Rx                DL:  8    d7 00 03 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.814603    ID:      782    S Rx                DL:  8    ff 44 01 15 01 08 06 ff     Channel: 0
[MASTER]: Timestamp: 1733813801.816758    ID:      7b1    S Rx                DL:  8    d7 00 04 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.824614    ID:      782    S Rx                DL:  8    ff 44 01 12 0a 08 05 ff     Channel: 0
[MASTER]: Timestamp: 1733813801.827547    ID:      7b1    S Rx                DL:  8    d6 00 00 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.834641    ID:      782    S Rx                DL:  8    ff ff ff ff ff ff ff ff     Channel: 0
[MASTER]: Timestamp: 1733813801.836304    ID:      7b1    S Rx                DL:  8    d5 00 01 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.844635    ID:      782    S Rx                DL:  8    ff ff ff ff ff ff ff ff     Channel: 0
[MASTER]: Timestamp: 1733813801.846110    ID:      7b1    S Rx                DL:  8    d4 00 00 00 01 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.854613    ID:      782    S Rx                DL:  8    ff ff ff ff ff ff ff ff     Channel: 0
[MASTER]: Timestamp: 1733813801.856292    ID:      7b1    S Rx                DL:  8    d3 00 00 00 00 01 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.864615    ID:      782    S Rx                DL:  8    ff ff ff ff ff ff ff ff     Channel: 0
[MASTER]: Timestamp: 1733813801.866491    ID:      7b1    S Rx                DL:  8    e2 00 00 00 00 00 00 00     Channel: 0
[SLAVE]:  Timestamp: 1733813801.874626    ID:      782    S Rx                DL:  8    ff ff ff ff ff ff ff ff     Channel: 0
[MASTER]: Timestamp: 1733813801.876772    ID:      7b1    S Rx                DL:  8    e1 ff 01 00 99 cb 34 34     Channel: 0
[SLAVE]:  Timestamp: 1733813801.884637    ID:      782    S Rx                DL:  8    ff ff ff ff ff ff ff ff     Channel: 0
[MASTER]: Timestamp: 1733813801.886267    ID:      7b1    S Rx                DL:  8    e0 00 00 00 00 00 01 06     Channel: 0
[SLAVE]:  Timestamp: 1733813801.894614    ID:      782    S Rx                DL:  8    ff ff ff ff ff ff ff ff     Channel: 0
[... cut ...]
2. My second question is whether there is a way to directly retrieve the values of the variables without having to export to XMRAW and then convert to CSV or Excel. Are the measured variable values stored in a list, dictionary, or something similar?
3. Additionally, could you specify which function you are using to decode DAQ/ShortUpload messages into actual values based on their hexadecimal format and data type, such as FLOAT32_IEEE, UBYTE, or UWORD? Does the library already provide a function that directly supports this? Please point it out if available. 

You are helping me a lot with my project. Thank you very much!

christoph2, 2024-12-11:
Looks much like that the recorder extension crashes -- this was an older issue., please make sure to use the latest pyXCP version, 
in any case run
`pip install -e .`
to rebuild the C++ extensions.

Coming to your second question: No, it is not required to record first and then do some post-processing; it's also
possible to process DAQ lists live, as they are received.

For an example s. [DaqToCsv](https://github.com/christoph2/pyxcp/blob/982f6ab595daa1ea2c6e0d7edc73535f9cfe6820/pyxcp/daq_stim/__init__.py#L201) class, you may use this as an starting point.
The interface of the base class has  three methods that must be overriden (C++ pure virtual methods under the hood):
```python
   def initialize(self):
       pass

   def on_daq_list(self, daq_list: int, ts0: int, ts1: int, payload: list):
      pass # Live processing here.

   def finalize(self):
       pass
```
`initialize` and `finalize` can be used to setup/close resources like files.

The parameters of `on_daq_list`  are:
- daq_list:   the number of the DAQ list received.
- ts0:  Timestamp generated py pyXCP.
- ts1:  Timestamp from DAQ list (if available)
- payload: a list of the converted measurement values.

The conversion routines are included in the fiile [unfolder.hpp](https://github.com/christoph2/pyxcp/blob/982f6ab595daa1ea2c6e0d7edc73535f9cfe6820/pyxcp/recorder/unfolder.hpp#L36).
They are used to convert the raw bytes to the desired datatype values (byte-order is taken into account), i.e. the last parameter in the measurements definition, e.g.:
```python
            ("swordCounter", 0x20414, 0, "I16"),
            ("sdwordCounter", 0x20418, 0, "I32"),
            ("channel1", 0x203F8, 0, "F64"),
            ("channel2", 0x20400, 0, "F64"),
            ("channel3", 0x20408, 0, "F64"),
```
Conversion is currently not available in the Python space, i.e. only raw byte values from `UPLOAD`/`SHORT_UPLOAD` ;
but a decent polling mode API could be added.

shirayr, 2024-12-15:
@PCT-VF can you share your config file(pyxcp_conf.py)?
Thanks!
iam using this code:
`import time
from pyxcp.cmdline import ArgumentParser
from pyxcp.daq_stim import DaqList, DaqRecorder
import logging

logging.basicConfig(level=logging.DEBUG)

# Define measurements directly
measurements = [
    ("COM_WindowPositionFL", 0x3434CA50, 0, "U8"),
    ("COM_WindowLockDSBBtnPress", 0x3434CA51, 0, "U8"),
    ("COM_WindowFRThermalProtect", 0x3434CA52, 0, "U8"),
    ("COM_WindowFRSuppressionWarn", 0x3434CA53, 0, "U8"),
]

logging.info("DAQ Measurements: %s", measurements)

daq_lists = [
    DaqList(
        "demo_measurement",
        0,    # EVENT_CHANNEL
        False,  # List is DAQ not STIM.
        True,   # Use DAQ timestamps if available.
        measurements,
    )
]

logging.info("Init DAQ Argument Parser")
ap = ArgumentParser(description="DAQ test")

daq_parser = DaqRecorder(daq_lists, "demo_measurement", 1)
logging.info("Initialized DAQ Recorder")

try:
    with ap.run(policy=daq_parser) as x:
        # logging.info("Connecting to ECU...")
        x.connect()
        # logging.info("Connected to ECU via XCP on CAN")
        if x.slaveProperties.optionalCommMode:
            x.getCommModeInfo()

        x.cond_unlock("DAQ")  # Unlock DAQ resource if locked.

        # logging.info("Setting up DAQ lists...")
        try:
            daq_parser.setup()
        except Exception as e:
            logging.error(f"Failed to setup DAQ lists: {e}")
        
        # logging.info("Starting DAQ...")
        daq_parser.start()

        time.sleep(60)  # Run for 60 seconds.

        # logging.info("Stopping DAQ...")
        daq_parser.stop()
        # logging.info("Finalizing DAQ lists.")
        x.disconnect()
except Exception as e:
    logging.error(f"An error occurred: {e}")`
failing on: daq_parser.setup()
Failed to setup DAQ lists: SystemExit(error_code=ERR_CMD_UNKNOWN, message='Could not proceed due to unhandled error (USE_A2L).')

how can i define it? and what if i dont want to read it from a2l file?
Thanks again for your support!!

PCT-VF, 2024-12-17:
This is my config in`pyxcp_conf.py`
```command
c = get_config()  # noqa
c.Transport.layer = 'CAN'
c.Transport.Can.interface = 'vector'
c.Transport.Can.can_id_master = 0x7B1  # Master CAN ID
c.Transport.Can.can_id_slave  = 0x782  # Slave CAN ID
c.Transport.Can.channel = 0  # CAN channel (e.g., 0 for Kvaser channel 0)
c.Transport.Can.bitrate = 500000  # CAN bitrate
c.Transport.Can.Vector.app_name = 'python-can'
```
In my code, I define measurement list directly instead of using A2L file, you can use it as a example.
```command
measurements = [
("COM_WindowPositionFL", 0x3434CA50, 0, "U8"),
("COM_WindowLockDSBBtnPress", 0x3434CA51, 0, "U8"),
("COM_WindowFRThermalProtect", 0x3434CA52, 0, "U8"),
("COM_WindowFRSuppressionWarn", 0x3434CA53, 0, "U8"),
]
```
I haven't ever seen your error, however, if you provide more infomation, maybe I can help you.

PCT-VF, 2024-12-27:
Hi @christoph2 

I am using your `pyXCP` library with the configuration file `pyxcp_conf.py`, which contains the following content:

```python
c = get_config()  # noqa
c.Transport.layer = 'CAN'
c.Transport.Can.interface = 'vector'
c.Transport.Can.can_id_master = 0x7B1  # Master CAN ID
c.Transport.Can.can_id_slave = 0x782  # Slave CAN ID
c.Transport.Can.channel = 0  # CAN channel (e.g., 0 for Kvaser channel 0)
c.Transport.Can.bitrate = 500000  # CAN bitrate
c.Transport.Can.Vector.app_name = 'python-can'
```
However, I am currently facing an issue where I need to modify the configuration while the program is running, so I don't want to load the configuration from an external file. Is there a way to declare the configuration directly by assigning values within the `main.py` file?

christoph2, 2024-12-27:
Well, pyXCP configuration uses [traitlets config](https://github.com/ipython/traitlets/tree/main/traitlets/config) under the hood, so 
your own configuration system should be based on this classes.

PCT-VF, 2024-12-27:
Can you give me an example of how to change the configuration for the channel, master, and slave, for example, switching from channel 1 to channel 2 in a main.py file? I understand that if I hardcode the configuration parameters into the `pyxcp_conf.py` file, I won't be able to change it dynamically.

christoph2, 2024-12-27:
Probably I don't understand your use-case...
But I think you certainly don't want to change channel while a measurement is running.
Two things come to mind:
1) `pyxcp_conf.py`  is a regular Python file, so it can contain arbitrary statements, functions, file and database access, ...
  so it's possible to put some dynamics in there, e.g.:
```python
if some_condition:
    c.Transport.Can.channel = 0
else:
   c.Transport.Can.channel = 1
```
2) I'm using `traitlets.config` as a blackbox, so I not really know whats going on under the hood -- but as far as I know the attributes should be writeable, you only need to figure out the data representation.
You may start your investigation with this code:
```python
from pyxcp.cmdline import ArgumentParser

ap = ArgumentParser(description="pyXCP hello world.")

with ap.run() as x:
    print(dir(x.transport.config))
```

PCT-VF, 2024-12-31:
@christoph2 . I am trying to use `slcan` for my CAN communication setup. I tested with `python-can` use following configuration:
`bus = can.interface.Bus(interface='slcan', channel='COM8', bitrate=500000)`
This configuration works as expected. However, when I attempt to configure the same in `pyxcp_conf.py` I face an issue. Here is my configuration file:
```code 
c = get_config()  # noqa
c.Transport.layer = 'CAN'
c.Transport.Can.interface = 'slcan'
c.Transport.Can.can_id_master = 0x7B1  # Master CAN ID
c.Transport.Can.can_id_slave  = 0x782  # Slave CAN ID
c.Transport.Can.channel = 'COM8'  # CAN channel (e.g., 0 for Kvaser channel 0)
```
When I run the `xcp-info.py` script, I encounter the following output:
```code 
PS C:\Tools\ACNI\DAL\utlis> python .\xcp-info.py
ldf is not supported
xls is not supported
DEBUG:root:Looking for pyxcp_conf in C:\Tools\ACNI\DAL\utlis
DEBUG:root:Loaded config file: C:\Tools\ACNI\DAL\utlis\pyxcp_conf.py
DEBUG:can.interface.detect_available_configs:interface "slcan" does not support detection of available configurations
WARNING:root:XCPonCAN - 'slcan' has no support for parameter 'fd'.
WARNING:root:XCPonCAN - 'slcan' has no support for parameter 'receive_own_messages'.
INFO:root:XCPonCAN - Interface-Type: 'slcan' Parameters: [('channel', 'COM8'), ('bitrate', 250000)]
INFO:root:XCPonCAN - Master-ID (Tx): 0x000007B1S -- Slave-ID (Rx): 0x00000782S
INFO:root:XCPonCAN - Using Interface: 'unknown'
INFO:root:XCPonCAN - Filters used: [{'can_id': 1922, 'can_mask': 2047, 'extended': False}]
INFO:root:XCPonCAN - State: BusState.ACTIVE
DEBUG:root:CONNECT
DEBUG:root:-> [ff 00 00 00 00 00 00 00]
DEBUG:root:CONNECT
DEBUG:root:-> [ff 00 00 00 00 00 00 00]
DEBUG:root:CONNECT
DEBUG:root:-> [ff 00 00 00 00 00 00 00]`
```
I would appreciate your guidance on resolving this issue.

christoph2, 2024-12-31:
At the first sight the bitrate:
You configuration is missing an explicit bitrate parameter, like
```python
c.Transport.Can.bitrate = 500000
```

so the default of 250kbs is used:
```shell
INFO     XCPonCAN - Interface-Type: 'slcan' Parameters: [('channel', 'COM8'), ('bitrate', 250000)]
```

PCT-VF, 2024-12-31:
Great, thank you for your support!

christoph2, 2025-01-07:
The `run()` method calls `create_application()` under the hood, which in turn constructs an application/configuration object -- if none exists, and yes this is sort of caching...

OK, as a first aid, I added a `reset_application()` function for this purpose:

```python
from pyxcp.cmdline import ArgumentParser, reset_application
```
Not really tested! I hope it works as expected.

shirayr, 2025-01-07:
Hi @christoph2 , am using this code:

pyxcp_main.py:
application = create_application(self.parser.options)
master = Master(
    application.transport.layer, config=application, policy=policy,
    transport_layer_interface=transport_layer_interface
)

and i have pyxcp_conf.py:
c = get_config()  # noqa
c.Transport.layer = 'ETH'
c.Transport.Eth.protocol = 'UDP'
c.Transport.Eth.bind_to_address = '....'
c.Transport.Eth.bind_to_port = ..
c.Transport.Eth.host = '....'
c.Transport.Eth.port = ..

can i change this row application = create_application(self.parser.options) by creating the config by myself in the pyxcp_main and send it to the Master as 'config' param? so that i wouldnt need the pyxcp_conf.

if so what should be the syntax?

PCT-VF, 2025-01-08:
> The `run()` method calls `create_application()` under the hood, which in turn constructs an application/configuration object -- if none exists, and yes this is sort of caching...
>
> OK, as a first [...]

I tested it, but it didn’t work. I noticed that on the very first call to `create_application()`, a `global application` already exists. It seems that some other function had initialized the application beforehand.
```code
Loading configuration from XCPTools_Configure.txt
[TEMP] XCP Channel: COM7 <- the initize COM
Return existing App: <pyxcp.config.PyXCP object at 0x00000147A2030410>
2025-01-08 09:13:57,038 - INFO - XCPonCAN - Interface-Type: 'slcan' Parameters: [('channel', 'COM7'), ('bitrate', 500000)]
2025-01-08 09:13:57,038 - INFO - XCPonCAN - Master-ID (Tx): 0x000007B1S -- Slave-ID (Rx): 0x00000782S
2025-01-08 09:13:57,038 - INFO - [XCP Module] INFO: Connecting to ECU . . .
2025-01-08 09:13:59,241 - INFO - XCPonCAN - Using Interface: 'unknown'
2025-01-08 09:13:59,256 - INFO - XCPonCAN - Filters used: [{'can_id': 1922, 'can_mask': 2047, 'extended': False}]
2025-01-08 09:13:59,269 - INFO - XCPonCAN - State: BusState.ACTIVE
2025-01-08 09:14:01,042 - INFO - [XCP Module] ERROR: Connection timeout
Reset APP
Loading configuration from XCPTools_Configure.txt
[TEMP] XCP Channel: COM8 <- I change channel in pyxcp_conf.py
Return existing App: <pyxcp.config.PyXCP object at 0x00000147A02551C0>
2025-01-08 09:16:40,659 - INFO - XCPonCAN - Interface-Type: 'slcan' Parameters: [('channel', 'COM7'), ('bitrate', 500000)]
2025-01-08 09:16:40,659 - INFO - XCPonCAN - Master-ID (Tx): 0x000007B1S -- Slave-ID (Rx): 0x00000782S
2025-01-08 09:16:40,660 - INFO - [XCP Module] INFO: Connecting to ECU . . .
2025-01-08 09:16:42,817 - INFO - XCPonCAN - Using Interface: 'unknown'
2025-01-08 09:16:42,848 - INFO - XCPonCAN - Filters used: [{'can_id': 1922, 'can_mask': 2047, 'extended': False}]
2025-01-08 09:16:42,870 - INFO - XCPonCAN - State: BusState.ACTIVE
2025-01-08 09:16:44,661 - INFO - [XCP Module] ERROR: Connection timeout
Reset APP
```

christoph2, 2025-01-08:
Hmm, currently the configration classes are `SingletonConfigurable`  - again a caching mechanism, I'll change them to `Configurable`.

christoph2, 2025-01-08:
@shirayr 
As I've written above:
> Well, pyXCP configuration uses [traitlets config](https://github.com/ipython/traitlets/tree/main/traitlets/config) under the hood, so your own configuration system should be based on this classes.

If you really want to bypass the configuration file, you must construct an compatible configuration object on your own...

P.S:
Is there a reason for
```python
c.Transport.Eth.bind_to_address = '....'
c.Transport.Eth.bind_to_port = ..
````
?

PCT-VF, 2025-01-08:
> Hmm, currently the configration classes are `SingletonConfigurable` - again a caching mechanism, I'll change them to `Configurable`.

I'll wait for your update. When there's an update in the latest version, please tag me here. Thank you so much!

christoph2, 2025-01-08:
@PCT-VF 
OK, update is done. The configuration object should not (hopefully) persist anymore after `reset_application()`.

PCT-VF, 2025-01-09:
Awesome! The configuration has indeed been updated properly upon reinitialization. Thank you so much for your support. I'm still continuing my project, and your library has been incredibly helpful to me.

christoph2, 2025-01-15:
@shirayr 
```python
getDaqInfodpi = self.getDaqProcessorInfo()
```
this means: at least the optional XCP service `DAQ_PROCESSOR_INFO` is missing; there may be even no DAQ support at all.
Run the script
```
xcp-info
```
to figure out (be sure to use the latest release).

shirayr, 2025-01-15:
@christoph2 thank you!
this is the output
Slave Properties:
=================
{'addressGranularity': EnumIntegerString.new(0, 'BYTE'),
 'byteOrder': EnumIntegerString.new(0, 'INTEL'),
 'bytesPerElement': 1,
 'maxCto': 255,
 'maxDto': 255,
 'maxWriteDaqMultipleElements': 31,
 'optionalCommMode': False,
 'pgmProcessor': {},
 'protocolLayerVersion': 1,
 'slaveBlockMode': False,
 'supportsCalpag': False,
 'supportsDaq': False,
 'supportsPgm': False,
 'supportsStim': False,
 'transportLayerVersion': 1,
 'transport_layer': 'ETH'}

Implemented IDs:
================

Protection Status
=================
    dbg   : False
    pgm   : False
    stim  : False
    daq   : False
    calpag: False
    
    were can i configure these properties?

christoph2, 2025-01-15:
@shirayr 
Well, the output of `xcp-info` reflects the configuration of your XCP slave -- re-configuration and compilation is required (if possible).
For instant results you may check [XCPlite](https://github.com/vectorgrp/xcplite) by Vector Informatik.

bsr656, 2025-03-24:
@christoph2 , I am trying to get the basic xcphello.py and I have this error . I already configured pyxcp_conf.py with the following parameters and I still have the error. Could you give me some hints to fix this?

c.Transport.Eth.host = "10.1.0.52"
c.Transport.Eth.port = 5555
c.Transport.Eth.protocol = 'UDP'

2025-03-24 15:54:42 INFO     XCPonEth - Connected to: 10.1.0.52:5555  local address: 10.1.0.47:47407 [AF_INET][SOCK_DGRAM]                                                                                                         eth.py:103
2025-03-24 15:54:44 WARNING  XcpTimeoutError -- Service: 'connect'                                                                                                                                                        errorhandler.py:371
                    INFO     XCPonEth - Connected to: 10.1.0.52:5555  local address: 10.1.0.47:47407 [AF_INET][SOCK_DGRAM]                                                                                                         eth.py:103
2025-03-24 15:54:46 WARNING  XcpTimeoutError -- Service: 'connect'                                                                                                                                                        errorhandler.py:371
                    ERROR    Traceback (most recent call last):                                                                                                                                                                 master.py:129
                               File "xcphello.py", line 30, in <module>
                                 x.connect()
                               File "venv/lib/python3.10/site-packages/pyxcp/master/errorhandler.py", line 437, in inner
                                 res = executor(inst, func, arguments)
                               File "venv/lib/python3.10/site-packages/pyxcp/master/errorhandler.py", line 364, in __call__
                                 res = handler.execute()
                               File "venv/lib/python3.10/site-packages/pyxcp/master/errorhandler.py", line 230, in execute
                                 return self.func(self.instance, *self.arguments.args, **self.arguments.kwargs)
                               File "venv/lib/python3.10/site-packages/pyxcp/master/master.py", line 176, in connect
                                 response = self.transport.request(types.Command.CONNECT, mode & 0xFF)
                               File "venv/lib/python3.10/site-packages/pyxcp/transport/base.py", line 260, in request
                                 return self._request_internal(cmd, False, *data)
                               File "venv/lib/python3.10/site-packages/pyxcp/transport/base.py", line 238, in
                             _request_internal
                                 self.send(frame)
                               File "venv/lib/python3.10/site-packages/pyxcp/transport/eth.py", line 207, in send
                                 self.sock.send(frame)
                             ConnectionRefusedError: [Errno 111] Connection refused

christoph2, 2025-06-17:
@GarethGZ 
Well, `CRC_OUT_OF_RANGE` is in this context not really an error (unless pyXCP is misbehaving), meaning the IDs 6, 7, 219, 220 are not supported.

GarethGZ, 2025-08-05:
> Well, `pyXCP` s basically about XCP and `pya2ldb` is about A2L... The interaction of these and other projects will be the task of [asamint](https://github.com/christoph2/asamint).
> 
> I've created a small example to show the basic procedure of running a DAQ list measurement (Tested with [XCPlite](https://github.com/vectorgrp/XCPlite)).
> 
> Note: pya2ldb uses SQLAlchemy under the hood, so arbitrary complex queries are possible, but for the sake of simplicity the example code selects all `MEASUREMENT`s.
> 
> Tip: There is a small command-line utility [xcp_fetch_a2l](https://github.com/christoph2/pyxcp/blob/master/pyxcp/scripts/xcp_fetch_a2l.py) that can be used to fetch the correspondending A2L file.
> 
> import time
> 
> from pyxcp.cmdline import ArgumentParser
> from pyxcp.daq_stim import DaqList, DaqRecorder
> 
> from pya2l import DB
> from pya2l import model
> 
> A2L_FILE = "C_Demo" # No extension -- .a2l is assumed.
> 
> db = DB()
> session = db.open_create(A2L_FILE)  # Import A2L or open existing .a2ldb
> 
> ASAM_TYPE_MAP = {
>     "UBYTE": "U8",
>     "SBYTE": "I8",
>     "UWORD": "U16",
>     "SWORD": "I16",
>     "ULONG": "U32",
>     "SLONG": "I32",
>     "A_UINT64": "U64",
>     "A_INT64": "I64",
>     "FLOAT16_IEEE": "F16",
>     "FLOAT32_IEEE": "F32",
>     "FLOAT64_IEEE": "F64",
> }        
>     
> measurements = []
> for m in session.query(model.Measurement).order_by(model.Measurement.name).all():
>     name = m.name
>     address = m.ecu_address.address
>     address_ext = m.ecu_address_extension.extension if m.ecu_address_extension else 0
>     asam_type = m.datatype
>     data_type = ASAM_TYPE_MAP.get(asam_type)
> 
>     # You may check here if the measurement should be used.
>     measurements.append((name, address, address_ext, data_type, ))
>     print(f"MEASUREMENT:  {name:<20s} address: 0x{address:08x}:{address_ext}  -- type: {asam_type!r}")
> 
> 
> ap = ArgumentParser(description="DAQ test")
> 
> EVENT_CHANNEL = 0
> 
> daq_lists = [
>     DaqList(
>         "demo_measurement",
>         EVENT_CHANNEL,
>         False, # List is DAQ not STIM.
>         True,  # Use DAQ timestamps if available.
>         measurements,
>     )
> ]
> 
> daq_parser = DaqRecorder(daq_lists, "demo_measurement", 1)
> 
> with ap.run(policy=daq_parser) as x:
>     x.connect()
>     if x.slaveProperties.optionalCommMode:
>         x.getCommModeInfo()
> 
>     x.cond_unlock("DAQ")  # DAQ resource is locked in many cases.
> 
>     print("setup DAQ lists.")
>     daq_parser.setup()
>     print("start DAQ lists.")
>     daq_parser.start()
> 
>     time.sleep(5 * 60)  # Run for 5 minutes.
> 
>     print("Stop DAQ....")
>     daq_parser.stop()
>     print("finalize DAQ lists.\n")
>     x.disconnect()

Hi @christoph2, maybe these are stupid questions. I logged the xmraw file and converted it to csv (xmraw-converter). The timestamp0 was always 0. Is it under expectation? Are DAQ log files always with two timestamps?

[demo_measurement.csv](https://github.com/user-attachments/files/21594009/demo_measurement.csv)

[demo_measurement.zip](https://github.com/user-attachments/files/21594040/demo_measurement.zip)

GarethGZ, 2025-08-08:
@christoph2 Hi, thank you for the reply. It will be perfect if timestamp0 can be turned off by a setting. Some software can only recognize the first column of the timestamp. e.g., ETAS MDA.

shirayr, 2025-09-16:
Hi @christoph2 , I would appreciate your help. 
I have been using Pyxcp=0.22.23 until today
And I initialized master:
`
Application = PyXCP()
Application.config_file = str(conf_file)
.....
Master = Master(
Application.transport.layer, **config=application**, policy=None,
Transport_layer_interface=None
)
`
I want to move on to pyxcp=0.23.9
But it does not accept the config as sent in the previous version
error:
AttributeError: 'dict' object has no attribute 'general'

**Config=application.config** does not work either
error:
AttributeError: 'LazyConfigValue' object has no attribute 
'disable_error_handling'

What should I send in this variable?
Thanks alot!!

shirayr, 2025-09-16:
xcp-profile convert -c pyxcp_conf.py -o new_config.py  
DONE!

but now getting:
Traceback (most recent call last):
  File "C:\Python310\lib\site-packages\pyxcp\master\errorhandler.py", line 415, in __call__
    res = handler.execute()
  File "C:\Python310\lib\site-packages\pyxcp\master\errorhandler.py", line 279, in execute
    return self.func(self.instance, *self.arguments.args, **self.arguments.kwargs)
  File "C:\Python310\lib\site-packages\pyxcp\master\master.py", line 181, in connect
    response = self.transport.request(types.Command.CONNECT, mode & 0xFF)
  File "C:\Python310\lib\site-packages\pyxcp\transport\base.py", line 276, in request
    return self._request_internal(cmd, False, *data)
  File "C:\Python310\lib\site-packages\pyxcp\transport\base.py", line 251, in _request_internal
    self.send(frame)
  File "C:\Python310\lib\site-packages\pyxcp\transport\eth.py", line 225, in send
    self.sock.send(frame)
OSError: [WinError 10057] A request to send or receive data was disallowed because the socket is not connected and (when sending on a datagram socket using a sendto call) no address was supplied

christoph2, 2025-09-16:
@shirayr 

`xcp-profile convert -c pyxcp_conf.py -o new_config.py`
This looks completely wrong. If you're using a pre-traitlets version, your config file is certainly not named pyxcp_conf.py...
Anyways, you may run

```shell
xcp-info -c old_config.toml 
```
to see what's going on. 
But an erroneous conversation of hostname/IP would be a completely new error that I can't reproduce.

closed 2026-02-14 by christoph2