christoph2/pyxcp #186: linux wheel files are not posted to PyPi since 0.22.0

ampz9, 2024-12-17:
similarly to this build: https://github.com/christoph2/pyxcp/actions/runs/12271561733/job/34238655559#step:5:465 since version 0.22.0 linux wheel files have not been posted to PyPi.

Some change in the cmake build file in 0.22.0 broke the Linux builds.

Is this in the roadmap to fix and post wheels for linux again to PyPi? Thanks!

christoph2, 2024-12-18:
No, the build process is not broken; the static Python library necessary to link C/C++ extensions is not available and 
`apt install libpython3-dev` does not work as expected.

ampz9, 2025-01-06:
is there a plan to post the Linux wheel files to PyPi?

pierreluctg, 2025-01-07:
Building in manylinux images is broken.

1. Because https://github.com/christoph2/pyxcp/blob/52befe1e306c07b2d73ea3f155735689e46563a6/CMakeLists.txt#L11 currently search for artifacts for Python 3 module and embedding developments. This is fixed by #190.
2. Because this project cannot build on manylinux2014 (GCC 10). This is fixed by #191.

ampz9, 2025-01-08:
Thanks @pierreluctg ... I'm closing this defect as resolved

closed 2025-01-08 by ampz9