christoph2/pyxcp #188: ImportError: DLL load failed while importing cpp_ext

wokequnimei, 2024-12-30:
When I used your latest pyxcp, I encountered the following problem:
[image]
The configuration I use is the previous pyxcp compatible version, as shown below：
[image]
I was excited when I learned about the new version of pyxcp, but was confused about how to use it. The new configuration system is how to configure 'can' to make 'Master' work.
Hope to get your reply, thank you very much！

wokequnimei, 2024-12-30:
The problem I encountered is the same as the one in the link (https://github.com/christoph2/pyxcp/discussions/173) . I tried the corresponding solution in the discussion, but it didn't work.

wokequnimei, 2024-12-30:
"D:\Program Files\Python310\python.exe" F:\New_pyxcp\pyxcp-master\pyxcp\scripts\xcp_examples.py 
Traceback (most recent call last):
  File "F:\New_pyxcp\pyxcp-master\pyxcp\scripts\xcp_examples.py", line 11, in <module>
    from pyxcp import console
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\__init__.py", line 11, in <module>
    from .master import Master  # noqa: F401, E402
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\__init__.py", line 9, in <module>
    from .master import Master  # noqa: F401
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\master.py", line 26, in <module>
    from pyxcp.daq_stim.stim import DaqEventInfo, Stim
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\daq_stim\__init__.py", line 9, in <module>
    from pyxcp.cpp_ext import DaqList
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\cpp_ext\__init__.py", line 1, in <module>
    from .cpp_ext import (  # noqa: F401
ImportError: DLL load failed while importing cpp_ext: 找不到指定的模块。

christoph2, 2024-12-30:
`pip install pyxcp` does not work?
Otherwise, if you cloned/downloaded  pyXCP, make sure to run `pip install -e .` 
to correctly build the project -- pyXCP uses a bunch of C++ extensions, the missing `cpp_ext` is one of them.

Note: You should use in the new configuration system, to make it work run:
```python
xcp-profile convert -c your_config_file.toml  -o pyxcp_conf.py
```

wokequnimei, 2024-12-30:
PS F:\New_pyxcp\pyxcp-master> pip310 install pyxcp
Requirement already satisfied: pyxcp in d:\program files\python310\lib\site-packages (0.22.10)
Requirement already satisfied: toml<0.11.0,>=0.10.2 in d:\program files\python310\lib\site-packages (from pyxcp) (0.10.2)
Requirement already satisfied: tomlkit<0.13.0,>=0.12.5 in d:\program files\python310\lib\site-packages (from pyxcp) (0.12.5)
Requirement already satisfied: rich<14.0.0,>=13.6.0 in d:\program files\python310\lib\site-packages (from pyxcp) (13.9.4)
Requirement already satisfied: uptime<4.0.0,>=3.0.1 in d:\program files\python310\lib\site-packages (from pyxcp) (3.0.1)
Requirement already satisfied: pyserial<4.0,>=3.5 in d:\program files\python310\lib\site-packages (from pyxcp) (3.5)
Requirement already satisfied: python-can<5.0.0,>=4.2.2 in d:\program files\python310\lib\site-packages (from pyxcp) (4.5.0)
Requirement already satisfied: pyusb<2.0.0,>=1.2.1 in d:\program files\python310\lib\site-packages (from pyxcp) (1.2.1)
Requirement already satisfied: construct<3.0.0,>=2.10.68 in d:\program files\python310\lib\site-packages (from pyxcp) (2.10.70)
Requirement already satisfied: line-profiler-pycharm<2.0.0,>=1.1.0 in d:\program files\python310\lib\site-packages (from pyxcp) (1.2.0)
Requirement already satisfied: bandit<2.0.0,>=1.7.8 in d:\program files\python310\lib\site-packages (from pyxcp) (1.8.0)
Requirement already satisfied: pytz<2025.0,>=2024.1 in d:\program files\python310\lib\site-packages (from pyxcp) (2024.2)
Requirement already satisfied: mako<2.0.0,>=1.2.4 in d:\program files\python310\lib\site-packages (from pyxcp) (1.3.8)
Requirement already satisfied: traitlets<=5.11.2 in d:\program files\python310\lib\site-packages (from pyxcp) (5.11.2)
Requirement already satisfied: stevedore>=1.20.0 in d:\program files\python310\lib\site-packages (from bandit<2.0.0,>=1.7.8->pyxcp) (5.4.0)
Requirement already satisfied: PyYAML>=5.3.1 in d:\program files\python310\lib\site-packages (from bandit<2.0.0,>=1.7.8->pyxcp) (6.0.2)
Requirement already satisfied: colorama>=0.3.9 in d:\program files\python310\lib\site-packages (from bandit<2.0.0,>=1.7.8->pyxcp) (0.4.6)
Requirement already satisfied: line-profiler in d:\program files\python310\lib\site-packages (from line-profiler-pycharm<2.0.0,>=1.1.0->pyxcp) (4.2.0)
Requirement already satisfied: MarkupSafe>=0.9.2 in d:\program files\python310\lib\site-packages (from mako<2.0.0,>=1.2.4->pyxcp) (3.0.2)
Requirement already satisfied: packaging>=23.1 in d:\program files\python310\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (24.2)
Requirement already satisfied: typing_extensions>=3.10.0.0 in d:\program files\python310\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (4.12.2)
Requirement already satisfied: wrapt~=1.10 in d:\program files\python310\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (1.17.0)
Requirement already satisfied: markdown-it-py>=2.2.0 in d:\program files\python310\lib\site-packages (from rich<14.0.0,>=13.6.0->pyxcp) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in d:\program files\python310\lib\site-packages (from rich<14.0.0,>=13.6.0->pyxcp) (2.18.0)
Requirement already satisfied: mdurl~=0.1 in d:\program files\python310\lib\site-packages (from markdown-it-py>=2.2.0->rich<14.0.0,>=13.6.0->pyxcp) (0.1.2)
Requirement already satisfied: pbr>=2.0.0 in d:\program files\python310\lib\site-packages (from stevedore>=1.20.0->bandit<2.0.0,>=1.7.8->pyxcp) (6.1.0)

[notice] A new release of pip is available: 23.0.1 -> 24.3.1
[notice] To update, run: D:\Program Files\Python310\python.exe -m pip install --upgrade pip
PS F:\New_pyxcp\pyxcp-master> xcp-profile convert -c your_config_file.toml  -o pyxcp_conf.py
Traceback (most recent call last):
  File "D:\Program Files\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Program Files\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\Program Files\Python310\Scripts\xcp-profile.exe\__main__.py", line 4, in <module>
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\__init__.py", line 11, in <module>
    from .master import Master  # noqa: F401, E402
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\__init__.py", line 9, in <module>
    from .master import Master  # noqa: F401
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\master.py", line 26, in <module>
    from pyxcp.daq_stim.stim import DaqEventInfo, Stim
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\daq_stim\__init__.py", line 9, in <module>
    from pyxcp.cpp_ext import DaqList
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\cpp_ext\__init__.py", line 1, in <module>
    from .cpp_ext import (  # noqa: F401
ImportError: DLL load failed while importing cpp_ext: 找不到指定的模块。

> `pip install pyxcp`不起作用？ 否则，如果您克隆/下载了 pyXCP，请确保运行`pip install -e .` 以正确构建项目 - pyXCP 使用了许多 C++ 扩展，缺少的`cpp_ext`就是其中之一。
> 
> 注意：您应该在新的配置系统中使用，以使其运行：
> 
> ```python
> xcp-profile convert -c your_config_file.toml  -o pyxcp_conf.py
> ```

wokequnimei, 2024-12-30:
I used 'pip install pyxcp', then run 'xcp-profile convert -c your_config_file.toml -o pyxcp_conf.py' but got an error

christoph2, 2024-12-30:
The old error persists or is there a new one?

wokequnimei, 2024-12-30:
> The old error persists or is there a new one?

The old error persists，same error message

christoph2, 2024-12-30:
It seems the former installation is still active in `site-packages`.
You may uninstall first:
`pip uninstall pyxcp`

wokequnimei, 2024-12-30:
PS F:\New_pyxcp\pyxcp-master> pip310 uninstall pyxcp
Found existing installation: pyxcp 0.22.10
Uninstalling pyxcp-0.22.10:
  Would remove:
    d:\program files\python310\lib\site-packages\pyxcp-0.22.10.dist-info\*
    d:\program files\python310\lib\site-packages\pyxcp\*
    d:\program files\python310\scripts\pyxcp-probe-can-drivers.exe
    d:\program files\python310\scripts\xcp-examples.exe
    d:\program files\python310\scripts\xcp-fetch-a2l.exe
    d:\program files\python310\scripts\xcp-id-scanner.exe
    d:\program files\python310\scripts\xcp-info.exe
    d:\program files\python310\scripts\xcp-profile.exe
Proceed (Y/n)? Y
  Successfully uninstalled pyxcp-0.22.10
PS F:\New_pyxcp\pyxcp-master> pip310 install pyxcp
Collecting pyxcp
  Using cached pyxcp-0.22.13-cp310-cp310-win_amd64.whl (1.3 MB)
Requirement already satisfied: pyserial<4.0,>=3.5 in d:\program files\python310\lib\site-packages (from pyxcp) (3.5)
Requirement already satisfied: mako<2.0.0,>=1.2.4 in d:\program files\python310\lib\site-packages (from pyxcp) (1.3.8)
Requirement already satisfied: python-can<5.0.0,>=4.2.2 in d:\program files\python310\lib\site-packages (from pyxcp) (4.5.0)
Requirement already satisfied: line-profiler-pycharm<2.0.0,>=1.1.0 in d:\program files\python310\lib\site-packages (from pyxcp) (1.2.0)
Requirement already satisfied: uptime<4.0.0,>=3.0.1 in d:\program files\python310\lib\site-packages (from pyxcp) (3.0.1)
Requirement already satisfied: pyusb<2.0.0,>=1.2.1 in d:\program files\python310\lib\site-packages (from pyxcp) (1.2.1)
Requirement already satisfied: toml<0.11.0,>=0.10.2 in d:\program files\python310\lib\site-packages (from pyxcp) (0.10.2)
Requirement already satisfied: construct<3.0.0,>=2.10.68 in d:\program files\python310\lib\site-packages (from pyxcp) (2.10.70)
Requirement already satisfied: rich<14.0.0,>=13.6.0 in d:\program files\python310\lib\site-packages (from pyxcp) (13.9.4)
Requirement already satisfied: bandit<2.0.0,>=1.7.8 in d:\program files\python310\lib\site-packages (from pyxcp) (1.8.0)
Requirement already satisfied: chardet<6.0.0,>=5.2.0 in d:\program files\python310\lib\site-packages (from pyxcp) (5.2.0)
Requirement already satisfied: traitlets<=5.11.2 in d:\program files\python310\lib\site-packages (from pyxcp) (5.11.2)
Requirement already satisfied: tomlkit<0.13.0,>=0.12.5 in d:\program files\python310\lib\site-packages (from pyxcp) (0.12.5)
Requirement already satisfied: pytz<2025.0,>=2024.1 in d:\program files\python310\lib\site-packages (from pyxcp) (2024.2)
Requirement already satisfied: colorama>=0.3.9 in d:\program files\python310\lib\site-packages (from bandit<2.0.0,>=1.7.8->pyxcp) (0.4.6)
Requirement already satisfied: PyYAML>=5.3.1 in d:\program files\python310\lib\site-packages (from bandit<2.0.0,>=1.7.8->pyxcp) (6.0.2)
Requirement already satisfied: line-profiler in d:\program files\python310\lib\site-packages (from line-profiler-pycharm<2.0.0,>=1.1.0->pyxcp) (4.2.0)
Requirement already satisfied: MarkupSafe>=0.9.2 in d:\program files\python310\lib\site-packages (from mako<2.0.0,>=1.2.4->pyxcp) (3.0.2)
Requirement already satisfied: packaging>=23.1 in d:\program files\python310\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (24.2)
Requirement already satisfied: wrapt~=1.10 in d:\program files\python310\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (1.17.0)
Requirement already satisfied: typing_extensions>=3.10.0.0 in d:\program files\python310\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (4.12.2)
Requirement already satisfied: markdown-it-py>=2.2.0 in d:\program files\python310\lib\site-packages (from rich<14.0.0,>=13.6.0->pyxcp) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in d:\program files\python310\lib\site-packages (from rich<14.0.0,>=13.6.0->pyxcp) (2.18.0)
Requirement already satisfied: mdurl~=0.1 in d:\program files\python310\lib\site-packages (from markdown-it-py>=2.2.0->rich<14.0.0,>=13.6.0->pyxcp) (0.1.2)
Requirement already satisfied: pbr>=2.0.0 in d:\program files\python310\lib\site-packages (from stevedore>=1.20.0->bandit<2.0.0,>=1.7.8->pyxcp) (6.1.0)
Installing collected packages: pyxcp
Successfully installed pyxcp-0.22.13

[notice] A new release of pip is available: 23.0.1 -> 24.3.1
[notice] To update, run: D:\Program Files\Python310\python.exe -m pip install --upgrade pip
PS F:\New_pyxcp\pyxcp-master> xcp-profile convert -c your_config_file.toml  -o pyxcp_conf.py
Traceback (most recent call last):
  File "D:\Program Files\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Program Files\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\Program Files\Python310\Scripts\xcp-profile.exe\__main__.py", line 4, in <module>
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\__init__.py", line 11, in <module>
    from .master import Master  # noqa: F401, E402
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\__init__.py", line 9, in <module>
    from .master import Master  # noqa: F401
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\master.py", line 26, in <module>
    from pyxcp.daq_stim.stim import DaqEventInfo, Stim
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\daq_stim\__init__.py", line 9, in <module>
    from pyxcp.cpp_ext import DaqList
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\cpp_ext\__init__.py", line 1, in <module>
    from .cpp_ext import (  # noqa: F401
ImportError: DLL load failed while importing cpp_ext: 找不到指定的模块。

**I uninstalled and installed again, but still got the same error**

christoph2, 2024-12-30:
OK, the binary distro (wheel) is used.
`D:\Program Files\Python310\lib\site-packages\pyxcp\pyxcp\cpp_ext\` should contain a `.pyd` file, something like
`cpp_ext.cp310-win_amd64.pyd`, if not, something went wrong while building the wheel, but in this case it would not be available on PyPI -- strange.
In any case, you should try the local build `pip install -e .`

PS.: `your_config_file.toml` is just a placeholder...

wokequnimei, 2024-12-30:
[image]
There is cpp_ext.cp310-win_amd64.pyd in D:\Program Files\Python310\Lib\site-packages\pyxcp\cpp_ext

christoph2, 2024-12-30:
OK, maybe Python 3.10 has a problem with relative imports, I'll try to fix it.

christoph2, 2024-12-30:
OK, v0.22.14 is now available, I hope it fixes the issue.

wokequnimei, 2024-12-31:
PS F:\New_pyxcp\pyxcp-master\pyxcp\examples> xcp-profile convert -c your_config_file.toml  -o pyxcp_conf.py
Traceback (most recent call last):
  File "D:\Program Files\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Program Files\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\Program Files\Python310\Scripts\xcp-profile.exe\__main__.py", line 4, in <module>
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\__init__.py", line 11, in <module>
    from .master import Master  # noqa: F401, E402
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\__init__.py", line 9, in <module>
    from .master import Master  # noqa: F401
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\master\master.py", line 26, in <module>
    from pyxcp.daq_stim.stim import DaqEventInfo, Stim
  File "D:\Program Files\Python310\lib\site-packages\pyxcp\daq_stim\__init__.py", line 9, in <module>
    from pyxcp.cpp_ext.cpp_ext import DaqList
ImportError: DLL load failed while importing cpp_ext: 找不到指定的模块。
**I have updated to v0.22.15, but still report the problem**

> OK, v0.22.14 is now available, I hope it fixes the issue.

christoph2, 2024-12-31:
Well, I think your Python installation is fooled up. 
Why are you using these strange runpy.py stuff and not the offical python.exe or the py launcher?

SaxElectronics, 2025-05-07:
I had the very same issue, it was related to missing c++ dlls.

- libstdc++-6.dll - The C++ standard library implementation
- libgcc_s_seh-1.dll - GCC's support library for exception handling
- libwinpthread-1.dll - POSIX thread support library

You have to copy these manually.

christoph2, 2026-03-01:
**Issue Analysis Complete - v0.28.0 Works Correctly ✅**

I've thoroughly tested the v0.28.0 Windows wheel (`pyxcp-0.28.0-cp313-cp313-win_amd64.whl`) in a clean virtual environment and **the import works perfectly**:

``powershell
python -m venv test_venv
.\test_venv\Scripts\Activate.ps1
pip install pyxcp==0.28.0
python -c "from pyxcp.cpp_ext.cpp_ext import DaqList; print('SUCCESS')"
# Output: SUCCESS ✅
``

**Root Cause of Original Issue:**
- This was reported with v0.22.10-0.22.13 (Dec 2024)
- @christoph2 fixed the relative import issue in **v0.22.14**
- The Windows wheels are built with **MSVC** (not MinGW), so no external DLL dependencies exist
- Current wheels bundle all necessary C++ extensions as `.pyd` files

**For Users Still Experiencing Issues:**

1. **Update to latest version:**
   ``bash
   pip uninstall pyxcp
   pip install --no-cache-dir pyxcp
   ``

2. **If installing from source**, ensure CMake is installed:
   ``bash
   pip install cmake
   pip install -e .
   ``

3. **Check Python version compatibility:**
   - pyxcp 0.28.0 supports Python 3.10-3.14
   - Ensure your Python matches a supported version

4. **Verify no conflicting installations:**
   ``bash
   pip list | findstr pyxcp
   # Should show only one pyxcp installation
   ``

**Note on MinGW DLLs (@SaxElectronics comment):**
The MinGW DLL issue only affects users who build from source with MinGW/GCC. Official PyPI wheels use MSVC and don't require external DLLs.

**Recommendation:** Close this issue as fixed in v0.22.14+ and current v0.28.0. If anyone still experiences import errors with v0.28.0+, please open a new issue with:
- Exact Python version (`python --version`)
- Installation method (pip/source)
- Full error traceback
- Output of `pip list | findstr pyxcp`

christoph2, 2026-03-01:
Closing - Fixed in v0.22.14+, verified working in v0.28.0