christoph2/pyxcp #189: problem installing pyxcp

wokequnimei, 2024-12-30:
### Discussed in https://github.com/christoph2/pyxcp/discussions/173

<div type='discussions-op-text'>

<sup>Originally posted by **tuttoaldoc** October  1, 2024</sup>
Hi,
I'm trying to install pyxcp but during the installation process I get an error. Could you give me any insight? I'm not very skilled with python. I'm usinng Python 3.9.6.
thanks
Michele

package Upgrade/Installation
Collecting pyxcp
  Downloading pyxcp-0.22.5.tar.gz (213 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting bandit<2.0.0,>=1.7.8 (from pyxcp)
  Downloading bandit-1.7.10-py3-none-any.whl.metadata (6.7 kB)
Collecting chardet<6.0.0,>=5.2.0 (from pyxcp)
  Downloading chardet-5.2.0-py3-none-any.whl.metadata (3.4 kB)
Collecting construct<3.0.0,>=2.10.68 (from pyxcp)
  Downloading construct-2.10.70-py3-none-any.whl.metadata (4.2 kB)
Collecting line-profiler-pycharm<2.0.0,>=1.1.0 (from pyxcp)
  Downloading line_profiler_pycharm-1.2.0-py3-none-any.whl.metadata (1.6 kB)
Requirement already satisfied: mako<2.0.0,>=1.2.4 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from pyxcp) (1.3.5)
Requirement already satisfied: pyserial<4.0,>=3.5 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from pyxcp) (3.5)
Collecting python-can<5.0.0,>=4.2.2 (from pyxcp)
  Downloading python_can-4.4.2-py3-none-any.whl.metadata (8.8 kB)
Collecting pytz<2025.0,>=2024.1 (from pyxcp)
  Downloading pytz-2024.2-py2.py3-none-any.whl.metadata (22 kB)
Collecting pyusb<2.0.0,>=1.2.1 (from pyxcp)
  Downloading pyusb-1.2.1-py3-none-any.whl.metadata (2.2 kB)
Collecting rich<14.0.0,>=13.6.0 (from pyxcp)
  Downloading rich-13.9.1-py3-none-any.whl.metadata (18 kB)
Requirement already satisfied: toml<0.11.0,>=0.10.2 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from pyxcp) (0.10.2)
Collecting tomlkit<0.13.0,>=0.12.5 (from pyxcp)
  Downloading tomlkit-0.12.5-py3-none-any.whl.metadata (2.7 kB)
Requirement already satisfied: traitlets<=5.11.2 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from pyxcp) (5.4.0)
Collecting uptime<4.0.0,>=3.0.1 (from pyxcp)
  Downloading uptime-3.0.1.tar.gz (6.6 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyYAML>=5.3.1 (from bandit<2.0.0,>=1.7.8->pyxcp)
  Downloading PyYAML-6.0.2-cp39-cp39-win_amd64.whl.metadata (2.1 kB)
Collecting stevedore>=1.20.0 (from bandit<2.0.0,>=1.7.8->pyxcp)
  Downloading stevedore-5.3.0-py3-none-any.whl.metadata (2.3 kB)
Requirement already satisfied: colorama>=0.3.9 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from bandit<2.0.0,>=1.7.8->pyxcp) (0.4.6)
Collecting line-profiler (from line-profiler-pycharm<2.0.0,>=1.1.0->pyxcp)
  Downloading line_profiler-4.1.3-cp39-cp39-win_amd64.whl.metadata (35 kB)
Requirement already satisfied: MarkupSafe>=0.9.2 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from mako<2.0.0,>=1.2.4->pyxcp) (2.1.1)
Requirement already satisfied: wrapt~=1.10 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (1.12.1)
Requirement already satisfied: packaging>=23.1 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (23.2)
Requirement already satisfied: typing-extensions>=3.10.0.0 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (4.8.0)
Requirement already satisfied: pywin32>=305 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from python-can<5.0.0,>=4.2.2->pyxcp) (305)
Collecting markdown-it-py>=2.2.0 (from rich<14.0.0,>=13.6.0->pyxcp)
  Downloading markdown_it_py-3.0.0-py3-none-any.whl.metadata (6.9 kB)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in c:\dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages (from rich<14.0.0,>=13.6.0->pyxcp) (2.16.1)
Collecting mdurl~=0.1 (from markdown-it-py>=2.2.0->rich<14.0.0,>=13.6.0->pyxcp)
  Downloading mdurl-0.1.2-py3-none-any.whl.metadata (1.6 kB)
Collecting pbr>=2.0.0 (from stevedore>=1.20.0->bandit<2.0.0,>=1.7.8->pyxcp)
  Downloading pbr-6.1.0-py2.py3-none-any.whl.metadata (3.4 kB)
Downloading bandit-1.7.10-py3-none-any.whl (130 kB)
Downloading chardet-5.2.0-py3-none-any.whl (199 kB)
Downloading construct-2.10.70-py3-none-any.whl (63 kB)
Downloading line_profiler_pycharm-1.2.0-py3-none-any.whl (4.8 kB)
Downloading python_can-4.4.2-py3-none-any.whl (265 kB)
Downloading pytz-2024.2-py2.py3-none-any.whl (508 kB)
Downloading pyusb-1.2.1-py3-none-any.whl (58 kB)
Downloading rich-13.9.1-py3-none-any.whl (242 kB)
[... 37 lines cut ...]
        File "C:\Dev\hot_upgrade\atp_core\socomecenv\socomecenv\lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
        File "C:\Users\MCO\AppData\Local\Temp\pip-build-env-uaeypfx8\overlay\Lib\site-packages\poetry\core\masonry\api.py", line 58, in build_wheel
          return WheelBuilder.make_in(
        File "C:\Users\MCO\AppData\Local\Temp\pip-build-env-uaeypfx8\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 88, in make_in
          wb.build(target_dir=directory)
        File "C:\Users\MCO\AppData\Local\Temp\pip-build-env-uaeypfx8\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 123, in build
          self._build(zip_file)
        File "C:\Users\MCO\AppData\Local\Temp\pip-build-env-uaeypfx8\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 172, in _build
          self._run_build_script(self._package.build_script)
        File "C:\Users\MCO\AppData\Local\Temp\pip-build-env-uaeypfx8\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 262, in _run_build_script
          subprocess.check_call([self.executable.as_posix(), build_script])
        File "C:\Program Files\Python\lib\subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['C:/Dev/hot_upgrade/atp_core/socomecenv/socomecenv/Scripts/python.exe', 'build_ext.py']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyxcp
  Building wheel for uptime (pyproject.toml) ... done
  Created wheel for uptime: filename=uptime-3.0.1-cp39-cp39-win_amd64.whl size=6375 sha256=8d535323defdbe468da15b42ba3ede2cb7037319bbeda68d0285424f616b246d
  Stored in directory: C:\Users\MCO\AppData\Local\Temp\pip-ephem-wheel-cache-cokso2ok\wheels\ff\d2\cd\e18d80e7974cad2a474c85ae336fedadc7d0b6ac7207dacf5c
Successfully built uptime
Failed to build pyxcp
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (pyxcp)</div>

closed 2024-12-30 by wokequnimei

mannyOaks, 2025-03-11:
How did you fix this?

Black8Mamba, 2025-07-05:
I encountered the same problem.

christoph2, 2025-07-08:
As mentioned in [discussion 173](https://github.com/christoph2/pyxcp/discussions/173), this means usually that [cmake](https://cmake.org) is not installed on your system.