christoph2/pyxcp #196: Cannot build or install in python 3.9 on linux

mannyOaks, 2025-01-08:
When installing latest version (0.22.18) via pip, pip downloads a .tar.gz and proceeds to build the wheel but fails.

Found that the linux build for python 3.9 is skipped [here](https://github.com/christoph2/pyxcp/blob/ceabe9ba74c7d77df6f68d0b4d9354f7fecfab45/pyproject.toml#L176)
[image]

mannyOaks, 2025-01-08:
Opened a PR that removes this skip, pls review https://github.com/christoph2/pyxcp/pull/197

mannyOaks, 2025-01-09:
I see that the PR I listed was merged, but builds still fail, I opened a new one to remove the manylinux overrides to see if this fixes the issue
https://github.com/christoph2/pyxcp/pull/198

christoph2, 2025-01-10:
@mannyOaks 
Good news!
[build and upload to PyPI](https://github.com/christoph2/pyxcp/actions/runs/12704018062) now runs to completion.

mannyOaks, 2025-01-10:
@christoph2 can confirm that I had no issues to install via pip install on multiple >3.9 versions

Thanks!