christoph2/pyxcp #199: when I Run XCP=0.22.22, test_master.py, it is ModuleNotFoundError: No module named 'pyxcp.cpp_ext.cpp_ext',how to Solve this problem?

XuHaobang, 2025-01-14:
[image]

Traceback (most recent call last):
  File "D:\Users\f88209c\Desktop\New folder (2)\最新XCP\pyxcp-master (9)\pyxcp-master\pyxcp\tests\test_master.py", line 7, in <module>
    from pyxcp import types
  File "D:\Users\f88209c\PycharmProjects\pythonProject2\venv\lib\site-packages\pyxcp\__init__.py", line 11, in <module>
    from .master import Master  # noqa: F401, E402
  File "D:\Users\f88209c\PycharmProjects\pythonProject2\venv\lib\site-packages\pyxcp\master\__init__.py", line 9, in <module>
    from .master import Master  # noqa: F401
  File "D:\Users\f88209c\PycharmProjects\pythonProject2\venv\lib\site-packages\pyxcp\master\master.py", line 27, in <module>
    from pyxcp.daq_stim.stim import DaqEventInfo, Stim
  File "D:\Users\f88209c\PycharmProjects\pythonProject2\venv\lib\site-packages\pyxcp\daq_stim\__init__.py", line 9, in <module>
    from pyxcp.cpp_ext.cpp_ext import DaqList
ModuleNotFoundError: No module named 'pyxcp.cpp_ext.cpp_ext'

christoph2, 2025-01-14:
The required build step is missing, refer to [this](https://github.com/christoph2/pyxcp/discussions/173#discussioncomment-10829380) answer.

christoph2, 2026-03-01:
Closing - Documented: source installs require CMake + build step. Use pip install pyxcp for pre-built wheels.