christoph2/pyxcp #205: pyxcp == 0.22.23 , shortUpload function often gets the Shortupload value as feedback from other messages, resulting in incorrect values returned

XuHaobang, 2025-01-17:
[image]

[image]

@christoph2   pyxcp == 0.22.23 , shortUpload function often gets the Shortupload value as feedback from other messages, resulting in incorrect values returned. Can I fix this? The version of pyxcp-0.21.11 I used before is OK, I hope you can help me, thank you very much ~

my pyxcp_conf.py  is :
c = get_config()  # noqa
# c.Transport.Can.Vector.serial = 529423  #CAN serial
c.Transport.Can.poll_interval = 0.001
c.Transport.Can.Vector.serial = 32634   #CAN serial
c.Transport.layer = 'CAN'
c.Transport.Can.padding_value = 0xAA
c.Transport.Can.use_default_listener = True
c.Transport.create_daq_timestamps = False
c.Transport.Can.Vector.rx_queue_size = 16384
c.Transport.Can.can_id_broadcast = 256
c.Transport.Can.interface = 'vector'
c.Transport.Can.max_dlc_required = True
c.Transport.Can.use_default_listener = True
c.Transport.Can.can_id_master = 0x580  # Master CAN ID
c.Transport.Can.can_id_slave = 0x581  # Slave CAN ID
c.Transport.Can.channel = 0  # CAN channel (e.g., 0 for Kvaser channel 0)
c.Transport.Can.bitrate = 500000  # CAN bitrate
c.Transport.Can.fd = True  # CAN bitrate
c.Transport.Can.receive_own_messages = True  # CAN bitrate
c.Transport.Can.data_bitrate = 2000000  # CAN bitrate
c.Transport.Can.sjw_abr = 8  # CAN bitrate
c.Transport.Can.tseg1_abr = 31  # CAN bitrate
c.Transport.Can.tseg2_abr = 8  # CAN bitrate
c.Transport.Can.sjw_dbr = 8  # CAN bitrate
c.Transport.Can.tseg1_dbr = 31  # CAN bitrate
c.Transport.Can.tseg2_dbr = 8  # CAN bitrate

christoph2, 2025-01-17:
The interesting thing here is the `-1`: 
The variable in question `Master.slaveProperties.maxCto` is set after a `CONNECT` service and one can expect a reasonable value for MAX_CTO, like 8 on CAN classic, but it seems to be zero here. A correct value is vital for block transfer, DAQ allocation and other things.

Could you run xcp-info and look for `maxCto`?

```python
{'addressGranularity': EnumIntegerString.new(0, 'BYTE'),
 'byteOrder': EnumIntegerString.new(0, 'INTEL'),
 'bytesPerElement': 1,
 'interleavedMode': False,
 'masterBlockMode': False,
 'maxBs': 255,
 'maxCto': 8,
 'maxDto': 8,
 'maxWriteDaqMultipleElements': 0,
 'minSt': 0,
 'optionalCommMode': True,
 'pgmProcessor': {},
 'protocolLayerVersion': 1,
 'queueSize': 0,
 'slaveBlockMode': True,
 'supportsCalpag': True,
 'supportsDaq': True,
 'supportsPgm': True,
 'supportsStim': False,
 'transportLayerVersion': 1,
 'transport_layer': 'CAN',
 'xcpDriverVersionNumber': 10}
```

And could you also put a print statement in your code?

```python
print(x.slaveProperties)
```

XuHaobang, 2025-01-21:
> The interesting thing here is the `-1`: The variable in question `Master.slaveProperties.maxCto` is set after a `CONNECT` service and one can expect a reasonable value for MAX_CTO, like 8 on CAN classic, but it seems to be zero here. A correct value is vital for block transfer, DAQ allocation and other things.
> 
> Could you run xcp-info and look for `maxCto`?
> 
> {'addressGranularity': EnumIntegerString.new(0, 'BYTE'),
>  'byteOrder': EnumIntegerString.new(0, 'INTEL'),
>  'bytesPerElement': 1,
>  'interleavedMode': False,
>  'masterBlockMode': False,
>  'maxBs': 255,
>  'maxCto': 8,
>  'maxDto': 8,
>  'maxWriteDaqMultipleElements': 0,
>  'minSt': 0,
>  'optionalCommMode': True,
>  'pgmProcessor': {},
>  'protocolLayerVersion': 1,
>  'queueSize': 0,
>  'slaveBlockMode': True,
>  'supportsCalpag': True,
>  'supportsDaq': True,
>  'supportsPgm': True,
>  'supportsStim': False,
>  'transportLayerVersion': 1,
>  'transport_layer': 'CAN',
>  'xcpDriverVersionNumber': 10}
> And could you also put a print statement in your code?
> 
> print(x.slaveProperties)

@christoph2   when I try to  print(x.slaveProperties) ,the result as below, can you tell me how to Work it out,thanks!
self.xm.slaveProperties = {'pgmProcessor': {}, 'transport_layer': 'CAN', 'byteOrder': EnumIntegerString.new(0, 'INTEL'), 'maxCto': 170, 'maxDto': 43690, 'supportsPgm': False, 'supportsStim': False, 'supportsDaq': False, 'supportsCalpag': False, 'slaveBlockMode': False, 'addressGranularity': EnumIntegerString.new(1, 'WORD'), 'protocolLayerVersion': 170, 'transportLayerVersion': 170, 'optionalCommMode': True, 'maxWriteDaqMultipleElements': 21, 'bytesPerElement': 2}

XuHaobang, 2025-01-21:
@christoph2 when I try to print(x.slaveProperties) ,the result as below, can you tell me how to Work it out,thanks!
**self.xm.slaveProperties = {'pgmProcessor': {}, 'transport_layer': 'CAN', 'byteOrder': EnumIntegerString.new(0, 'INTEL'), 'maxCto': 170, 'maxDto': 43690, 'supportsPgm': False, 'supportsStim': False, 'supportsDaq': False, 'supportsCalpag': False, 'slaveBlockMode': False, 'addressGranularity': EnumIntegerString.new(1, 'WORD'), 'protocolLayerVersion': 170, 'transportLayerVersion': 170, 'optionalCommMode': True, 'maxWriteDaqMultipleElements': 21, 'bytesPerElement': 2}**

christoph2, 2025-01-21:
Well, looks really messed up!

`maxCto` should be 8 on CAN classic and max. 64 on CAN-FD -- interestingly the value 170 repeats (protocolLayerVersion, transportLayerVersion).
And `xcp-info` gives the same values?
If available, you may check the configuration file of your XCP slave.

XuHaobang, 2025-01-21:
> Well, looks really messed up!
> 
> `maxCto` should be 8 on CAN classic and max. 64 on CAN-FD -- interestingly the value 170 repeats (protocolLayerVersion, transportLayerVersion). And `xcp-info` gives the same values? If available, you may check the configuration file of your XCP slave.

Hi thanks,  when I run the XCP_info.py,Slave Properties as below, if my pyxcp_conf.py  config is fault?  my  pyxcp_conf.py  is below:

 
[image]  

[image]

christoph2, 2025-01-21:
OK, you may turn `max_dlc_required` to `False`, because padding is not expected to work under this conditions.
Also note: the paramter of `shortUpload` and similar functions is the number of elements, not bytes!
In your case the element-size (addressGranularity) is WORD, i.e. you need to divide your `Characteristic_size_in_bytes`  by two.

XuHaobang, 2025-01-22:
@christoph2 when I turn max_dlc_required to False，Unfortunately, although no error was reported, the communication was abnormal. Xcp-info was incorrectly read from the beginning, which should be a bug, but I don't know where caused. I am currently using the old XCP version =pyxcp-0.21.11, and the parameters of xcp-info read out are all normal. max-cto = 8, and the value read by shortload is also correct. If you have conditions, you can test your latest version. It is really not possible. I think it is not the problem of the configuration file. They are all the same without any difference. I have checked them completely. If there is an environment on your side, you can Debug it.

christoph2, 2026-02-15:
## ✅ Fixed in commits 5c91c17 and c559b38

The root cause was **NOT a race condition** but rather **CAN padding being interpreted as data**.

### Root Cause
When using CAN-FD with max_dlc_required=True and padding_value=0xAA:
- CAN frames are padded to maximum DLC (64 bytes for CAN-FD)
- process_response() in ase.py was appending the **entire padded buffer** to esQueue
- Parsers then read padding bytes as valid data
- Result: 0xAA (170 decimal) appearing in parsed values

**Your reported symptoms:**
- maxCto: 170 (should be 8) ← 0xAA
- maxDto: 43690 (should be 8) ← 0xAAAA (two padding bytes)
- protocolLayerVersion: 170 ← 0xAA

### Solution
Modified process_response() to trim responses to their **actual length** before queueing:
\\\python
self.resQueue.append(response[:length])  # Was: response
\\\

Applied to all response types:
- CMD responses (PID 0xFE/0xFF)
- EVENT frames (0xFD)
- SERV frames (0xFC)
- DAQ frames

### Changes
- **pyxcp/transport/base.py**: Lines 392, 395, 398, 400, 403, 431 - all now use esponse[:length]
- **pyxcp/tests/test_issue_205_padding.py**: 4 new regression tests validating padding removal

### Testing
- ✅ All 312 tests pass (4 skipped)
- ✅ New tests validate padding (0x00, 0xAA, 0xFF, 0xCC) is correctly stripped
- ✅ Tests cover CONNECT responses, SHORT_UPLOAD, and DAQ frames

**Please test with your ECU configuration** (CAN-FD + max_dlc_required + padding) and confirm the fix works!

closed 2026-02-15 by christoph2