christoph2/pyxcp #206: Issue when writing with shortDownload on linux vs windows

mannyOaks, 2025-01-17:
Hi, it's me again xd

I have recently found an issue when writing through shortDownload, on windows it's working fine, but on linux it's writing 0:

Linux:
[image]

[image]

Windows:
[image]

[image]

Just by the looks of it, the frame is adding 4 0's at the end of the data on linux and not on windows
Any idea what might be happening?

mannyOaks, 2025-01-17:
Linux: python 3.9.17
Windows: python 3.9.13

christoph2, 2025-01-21:
Yes, in case of Linux the buffer is padded to 8 bytes.

But neither the length nor the data are modified by `shortDownload`:
```python
    def shortDownload(self, address, addressExt, data):
        length = len(data)
        addr = self.DWORD_pack(address)
        return self.transport.request(types.Command.SHORT_DOWNLOAD, length, 0, addressExt, *addr, *data)
```
There are `max_dlc_required` and `padding_value` options, but they only apply to CAN.

closed 2025-01-31 by mannyOaks