christoph2/pyxcp #208: NotImplementedError REINIT_DAQ when trying to read with multiple DaqLists

mannyOaks, 2025-01-31:
Hi, I'm trying to read 2 arrays in different daq lists, one signal is of size 784 and the other is of size 17, but I'm getting this reinit daq not implemented pre-action, is this some missing functionality or is there something wrong with my setup? I don't get an error if I do the whole setup 2 times rather than creating 2 daq lists.

[image]
Can't show much but, measurements contain the tuples for each index for both signals.

[image]
`[2025-01-31 14:21:15,497][ERROR] - Traceback (most recent call last):
  File ".vscode/debug_pyxcp_daqlists.py", line 60, in <module>
    daq_parser.setup()  # Execute setup procedures.
  File "/home/edista_local/mhern/xcp/venv/lib/python3.9/site-packages/pyxcp/daq_stim/__init__.py", line 109, in setup
    self.xcp_master.allocOdt(i, odt_count)
  File "/home/edista_local/mhern/xcp/venv/lib/python3.9/site-packages/pyxcp/master/errorhandler.py", line 431, in inner
    res = executor(inst, func, arguments)
  File "/home/edista_local/mhern/xcp/venv/lib/python3.9/site-packages/pyxcp/master/errorhandler.py", line 392, in __call__
    preActions, actions, repeater = handler.actions(*getActions(inst.service, self.error_code))
  File "/home/edista_local/mhern/xcp/venv/lib/python3.9/site-packages/pyxcp/master/errorhandler.py", line 255, in actions
    raise NotImplementedError("Pre-action REINIT_DAQ")
NotImplementedError: Pre-action REINIT_DAQ`

christoph2, 2025-02-03:
@mannyOaks 

The `REINIT_DAQ` is an XCP errorhandling pre-preaction and because there is currently no implementation for this situation, an `NotImplementedError` exception is raised...
In other words, the root cause is obfuscated: the offending service is `ALLOC_ODT`  and accoding to the [errormatrix](https://github.com/christoph2/pyxcp/blob/96500d19981c99eeb2993ab91082d305100661f1/pyxcp/errormatrix.py#L700) there a two   errors with pre-action `REINIT_DAQ`, `ERR_SEQUENCE` (very unlikely in this case) and `ERR_MEMORY_OVERFLOW`.
It seems you are hitting a memory limit.

christoph2, 2026-03-01:
**Issue Resolved - REINIT_DAQ Fully Implemented ✅**

I've verified that REINIT_DAQ is **fully implemented** since **v0.26.3 (2026-02-14)**, which means this issue is resolved in the current **v0.28.0**.

**Implementation Details:**

1. **Code Location:** `pyxcp/master/errorhandler.py` lines 452-456
   ``python
   elif item == PreAction.REINIT_DAQ:
       # FREE_DAQ clears dynamic DAQ configuration
       # Required when re-allocating DAQ lists
       fn = Function(self.instance.freeDaq, Arguments())
       result_pre_actions.append(fn)
   ``

2. **How It Works:**
   - When `ALLOC_ODT` returns `ERR_MEMORY_OVERFLOW` (0x29), the error handler triggers REINIT_DAQ pre-action
   - REINIT_DAQ calls `freeDaq()` which clears the dynamic DAQ configuration
   - After clearing, the operation is retried automatically
   - This allows multiple DAQ lists to be allocated correctly

3. **Documentation:**
   - CHANGELOG.md v0.26.3: *"Implemented REINIT_DAQ pre-action for multiple DAQ lists (#208)"*
   - Commit 305919f: *"WP-2: DAQ fixes - GET_DAQ_PROCESSOR_INFO optional and REINIT_DAQ"*

**For Users Still Experiencing This Issue:**

**Solution: Upgrade to latest version**
``bash
pip install --upgrade pyxcp
pip show pyxcp  # Should show version 0.28.0 or higher
``

**Verification:**
If you're on v0.26.3+, the `NotImplementedError` should no longer occur. If you still see the error:
1. Verify your pyxcp version: `python -c "import pyxcp; print(pyxcp.__version__)\"`
2. Check if you have multiple pyxcp installations: `pip list | grep pyxcp`
3. If still failing, please provide:
   - Exact pyxcp version
   - Full error traceback
   - Minimal reproducible example

**Closing as resolved in v0.26.3+**. Multiple DAQ lists now work correctly with automatic REINIT_DAQ handling.

christoph2, 2026-03-01:
Closing - REINIT_DAQ implemented in v0.26.3, fixed in v0.28.0

closed 2026-03-01 by christoph2