christoph2/pyxcp #209: Master.close() delay

fross, 2025-02-14:
Hi,

I'm making use of pyxcp and [udsoncan](https://github.com/pylessard/python-udsoncan) in unit tests.

**Question:**
Is there a way (or configuration) that can be used to reduce the time it takes for closing `Master`?

**Connect + disconnect using IXXAT (pyxcp):**
```
2025-02-14 13:05:12,924 - MyLogger  - Connecting...
2025-02-14 13:05:14,434 - can.ixxat  - CAN Filters: None
2025-02-14 13:05:14,434 - can.ixxat  - Searching for first available device
2025-02-14 13:05:14,520 - can.ixxat  - Using unique HW ID b'100023418'
2025-02-14 13:05:14,520 - can.ixxat  - Initializing channel 0 in shared mode, 16 rx buffers, 16 tx buffers
2025-02-14 13:05:14,531 - can.ixxat  - Initializing control 0 bitrate 1000000
2025-02-14 13:05:14,565 - MyLogger  - Connected
2025-02-14 13:05:14,565 - MyLogger  - Disconnecting...
2025-02-14 13:05:14,565 - MyLogger  - Master disconnecting...
2025-02-14 13:05:14,575 - MyLogger  - Master disconnected
2025-02-14 13:05:14,575 - MyLogger  - Master closing...   ***** takes ~5 seconds *****
2025-02-14 13:05:20,593 - MyLogger  - Master closed
2025-02-14 13:05:20,594 - MyLogger  - Disconnected
```

**Connect + disconnect using IXXAT (udsoncan):**
```
2025-02-14 13:05:43,039 - MyLogger  - Connecting...
2025-02-14 13:05:43,055 - can.ixxat  - CAN Filters: None
2025-02-14 13:05:43,055 - can.ixxat  - Searching for first available device
2025-02-14 13:05:43,123 - can.ixxat  - Using unique HW ID b'100023418'
2025-02-14 13:05:43,123 - can.ixxat  - Initializing channel 0 in shared mode, 16 rx buffers, 16 tx buffers
2025-02-14 13:05:43,123 - can.ixxat  - Initializing control 0 bitrate 1000000
2025-02-14 13:05:43,127 - Connection  - Connection opened
2025-02-14 13:05:43,127 - MyLogger  - Connected
2025-02-14 13:05:43,128 - MyLogger  - Disconnecting...
2025-02-14 13:05:43,128 - MyLogger  - UDS client closing...
2025-02-14 13:05:43,128 - Connection  - Connection closed
2025-02-14 13:05:43,128 - MyLogger  - UDS client closed
2025-02-14 13:05:43,128 - MyLogger  - CAN bus shutting down...
2025-02-14 13:05:44,145 - MyLogger  - CAN bus shutdown
2025-02-14 13:05:44,145 - MyLogger  - Disconnected
```

I understand this may simply be due to protocol differences between XCP and UDS. My test suite is quite exhaustive and reducing the disconnection delay of the XCP master would speed things up tremendously.

Thanks!

SaxElectronics, 2025-05-03:
Hi,
I have the very same observation (my UDS initialization+connection is much faster than the pyxcp initialization+connection), even when the xcp master does not receive a message from the pyxcp hangs and waits forever for a message. this blocks the usb port completely.

christoph2, 2025-05-05:
@fross 
@SaxElectronics 

A "forgotten" feature just came to my mind:

It is possible to reuse an existing connection, for instance a `canlib.Bus` object (pyxcp has no ownership, so there is hopefully less interference).
For this use-case there is a `transport_layer_interface` argument:
```
ap = ArgumentParser(description="")

# `can_bus` object initialized elsewhere.

with ap.run(transport_layer_interface=can_bus) as x:
    x.connect()
    .
    .
    .
````