christoph2/pyxcp #211: Using `pyxcp` as a library

akermu, 2025-02-28:
Thank you very much, for bringing `xcp` to the python world!

We want to use this package as a library and we have currently two issues:

* There's currently unconditionally pretty printing installed: https://github.com/christoph2/pyxcp/blob/36843fd640b180dd9077123945b5e5be08a1b694/pyxcp/__init__.py#L9-L16
* It is very difficult to make a `PyXCP` object programmatically as it expects to read from a file (can be worked around)

I would provide patches for both of these, if you're interested?

fross, 2025-05-03:
We're also interested in using this as a library.

We haven't spend much time on it yet, but the third point you are mentioning is problematic for us too.

We are still using 0.21 since `Master` can be configured using a simple dictionary.

SaxElectronics, 2025-05-11:
@akermu did you manage to use pyxcp as a library? I am having hard times initializing the drivers without the commandline tool.
I would like to have more control over what is going on under the hood to use it really as a library. All the initialization stuff is pretty "hidden".

fross, 2025-05-11:
@SaxElectronics I suspect it's only a matter of updating the configuration of `Master` from a dictionary to a `Config` object.

I gave it a shot last week, but did not have the time to complete the configuration (yet). I'll come back to it eventually. The [legacy module](https://github.com/christoph2/pyxcp/blob/master/pyxcp/config/legacy.py) is a good starting point to dig in this.

e.g.
```
    from traitlets.config import Config

    [...]

    general_config = Config()

    transport_config = Config()
    transport_config.timeout = 5.0

    can_config = Config
    can_config.bitrate = 500_000
    can_config.can_id_master = 257
    can_config.can_id_slave = 258
    can_config.channel = "0"
    can_config.daq_identifier = []
    can_config.data_bitrate = 500_000
    can_config.fd = False
    can_config.interface = "Kvaser"
    can_config.max_dlc_required = False
    can_config.use_default_listener = True

    config = Config()
    config.general = general_config
    config.transport = transport_config
    config.transport.can = can_config

    master = Master("CAN", config)
```

SaxElectronics, 2025-05-11:
> @SaxElectronics I suspect it's only a matter of updating the configuration of `Master` from a dictionary to a `Config` object.
>
> I gave it a shot last week, but did not have the time to complete the [...]

Did you find a way how to separately initialize the bus and pass it to the master. I think some patching of Master would be required as connect command initializes also the bus?

fross, 2025-05-11:
I have not.

The device I'm using has a dedicated port for XCP communication. It is not
a problem if pyxcp manages the bus since it is not shared.

SaxElectronics, 2025-05-11:
Sure it is possible and I am confident that "I'll figure it out too"... at some point. The problem is the time spent on it.
I thought in 1-2 days the update will be over, now almost a week on it. This is a bit frustrating.

akermu, 2025-05-12:
@SaxElectronics nice 👍 Your solution is actually cleaner, I just use a dummy config file:

```
        empty_config_file = NamedTemporaryFile(delete=False, suffix=".py")
        empty_config_file.close()

        config = PyXCP()
        config.config_file = empty_config_file.name
        config._read_configuration(config.config_file)
```

SaxElectronics, 2025-05-13:
Hi guys,

I have a small but important update for you, perhaps usefull for some of you. I have been testing my scripts with direct configuration and also observed an inconcistency while executing in different pyxcp versions. I managed to track this down to this change causing it:

[image]

This is a key difference, if you are working without a context managet you need to call transport.connect() before calling master.connect()! Otherwise your code will fail with an error that the transport layer instance instance is not available:

```
INFO     Connecting...
ERROR    Test failed: 'PythonCanWrapper' object has no attribute 'can_interface'
INFO     Test complete
```

@christoph2 introduced this fix to enable connecting and disconnecting from the slave in the very same session.

So correct order would to instanciate a connection (currently) to slave and bus would be:
```
master.transport.connect()
logger.info("Connected successfully")
response = master.connect()
logger.info(f"Connection successful! Response: {response}")
```
I think properly defined APIs in the master class with clear error handling would be very helpful for everyone. 
As the normal user, who is not that deep in to the library has difficulties understanding the error messages. 

1. API for connecting and disconnecting to the BUS
2. API for connecting and disconnect ion to the slave 1) is (of course) required for 2)
 proper error handlign can inform the user what he did wrong, if he decides to connect to the slave without a transport layer connection estableshed first.

SaxElectronics, 2025-05-21:
@fross @akermu if you are interested in the examples there are added here in my fork:
https://github.com/SaxElectronics/pyxcp/tree/master/pyxcp/examples

I can try to bring them in the upstream/master if @christoph2 wants them.