christoph2/pyxcp #223: Migrating to the latest pyxcp with no CanInterfaceBase class

still-learnin, 2025-07-16:
I am currently using pyxcp 21.9 in a Python 3.9 (and some projects in a 3.11) environment. I wanted to update to the latest as these are quite outdated now.

I updated to Python 3.13.5 which appears to be the latest for Windows. However if I then try and install the 21.9 pyxcp package into this environment (I use pycharm env and its interpreters) I get the following error. I could only find that others have said this is related to early release Python versions which this is not.

File "C:\Users\xxxxxxxx\AppData\Local\Temp\pip-build-env-jh4jizq4\overlay\Lib\site-packages\pdm\pep517\wheel.py", line 214, in tag
      tag in supported_tags
  AssertionError: would build wheel with unsupported tag ('cp313', 'cp313_win_amd64', 'win_amd64')
  [end of output]

So then I tried updating the pyxcp version in my 3.9 environment to 22.32, which installs okay but gives me the following error when I try and run my test case:

ImportError: cannot import name 'CanInterfaceBase' from 'pyxcp.transport.can' 
(C:\Users\xxxxxxxx\env\lib\site-packages\pyxcp\transport\can.py)

I see that there have been some changes in this area and the way that the interface is setup. It was a long time ago when I first set this up and I am wondering if there is an easy or recommended way to migrate to the new setup. My current use of that class is shown in the attached file. Any pointers much appreciated.

[XCP_copy.txt](https://github.com/user-attachments/files/21264514/XCP_copy.txt)

still-learnin, 2025-07-16:
Perhaps I should add that if I could solve the first issue then there is not such an urgent need to update my pyxcp, but I would like to migrate both eventually.

christoph2, 2025-07-17:
@still-learnin 
Yes, there is no `CanInterfaceBase` class anymore. But some time ago, there was a feature request, re-use an existing CAN bus object (e.g. to run XCP and UDSonCAN in paralell) -- this is now possible with the `transport_layer_interface`  argument:

```python    
import can

from pyxcp.cmdline import ArgumentParser

can_if = can.Bus(interface="kvaser", channel="0", fd=False, bitrate=500000)

ap = ArgumentParser(description="external interface test")

with ap.run(transport_layer_interface=can_if) as x:
	x.connect()	
	x.disconnect()
```

And thanks to Pythons duck-typing nature, it is possible to supply an arbitrary object, as long as the interface is compatible -- maybe I should re-introduce `CanInterfaceBase` as an abstract base class.

- The `unsupported tag` error is rather random and not specific to pyXCP (besides other versions I'm using 3.13.3 with no problems) -- updating the `wheel` package may help, and if you have installed multiple Pythons on Windows, be sure to use the `py` frontend, e.g. `py -V:3.13`
 
- While looking at your code I noticed some other things:
  - the `XCPResponseMonitor` functionality should replaced with a [policy](https://github.com/christoph2/pyxcp/blob/e38d5a2ea78e4d3d03f41038a2695f6519c5d22c/pyxcp/transport/base.py#L21), this mechanism is well integrated into pyXCP.
  - It isn't necassary to implement calibration functions, to [asamint project](https://github.com/christoph2/asamint) makes good progress; I'm optimistic that there is a useable version in two months.

still-learnin, 2025-08-13:
@christoph2 Thank you for the detailed response. This is a background task, hence the delay, but I have finally had a chance to try and progress this further.

Previously, I made good use of the 'CanInterfaceBase' class since the 'transmit' method allowed me to send the message on the correct CAN ID, which in my test application is dynamically configurable (both CMD and BCAST IDs - and on the ECUs themselves). So I can see the 'receive_callback' can potentially be replaced with the policies but is there an easy workaround for the 'transmit' part?

Secondly, although I was able to make use of the 'transport_layer_interface' parameter I still seemed to require a minimum configuration in my .toml file: 
TRANSPORT = "CAN"
CAN_DRIVER = "Vector"
Is this correct?
Previously my XCP class/driver was CAN interface agnostic, at least I could happily use both Vector and PCAN interfaces without the XCP class knowing anything about it. I was hoping I might be able to use an essentially empty .toml file, given that I supplied the 'transport_layer_interface'.

christoph2, 2025-08-17:
@still-learnin 
Hello,

[CanInterfaceBase](https://github.com/christoph2/pyxcp/blob/a09e6b628b4380f70f1a7818cac9032a42d89903/pyxcp/transport/can.py#L522) is back!

The [xcp_user_supplied_driver.py](https://github.com/christoph2/pyxcp/blob/master/pyxcp/examples/xcp_user_supplied_driver.py) example was adapted to the current release and [here](https://github.com/christoph2/pyxcp/blob/master/docs/custom_can_driver.md) you find some documentation on custom CAN interfaces.

Regarding configuration for custom CAN interfaces, the relevant entries are:
```
c.Transport.layer = 'CAN'
c.Transport.Can.interface="custom"
```