christoph2/pyxcp #230: Do DAQ( XCP_on_CAN) with a ECU that doesn't support GET_DAQ_PROCESSOR_INFO

GarethGZ, 2025-08-14:
@christoph2 
Hi, I tested your example in [https://github.com/christoph2/pyxcp/issues/184](https://github.com/christoph2/pyxcp/issues/184) with a real ECU. The ECU doesn't support GET_DAQ_PROCESSOR_INFO, so the error was thrown below. Please give some advice.

[PyXCP] XCPonCAN - Filters used: [{'can_id': 2033, 'can_mask': 2047, 'extended': False}]
[PyXCP] XCPonCAN - State: BusState.ACTIVE
           Level 9  Received: Timestamp: 1755157627.265973    ID:      7f1    S Rx                DL:  8    ff 05 c0 08  bus.py:131
                    08 00 01 01     Channel: 0                                                                                     
           Level 9  Received: Timestamp: 1755157627.269473    ID:      7f1    S Rx                DL:  8    ff 00 01 00  bus.py:131
                    2b 00 00 60     Channel: 0                                                                                     
           Level 9  Received: Timestamp: 1755157627.274373    ID:      7f1    S Rx                DL:  6    ff 00 00 00  bus.py:131
                    00 00           Channel: 0                                                                                     
setup DAQ lists.
           Level 9  Received: Timestamp: 1755157627.278113    ID:      7f1    S Rx                DL:  2    fe 20        bus.py:131
                    Channel: 0                                                                                                     
[PyXCP] ERROR | Traceback (most recent call last):
  File "d:\Programs\Python\Python313\Lib\site-packages\pyxcp\daq_stim\__init__.py", line 47, in setup
    self.daq_info = self.xcp_master.getDaqInfo(include_event_lists=False) # by_GZ
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\Programs\Python\Python313\Lib\site-packages\pyxcp\master\master.py", line 1675, in getDaqInfo
    dpi = self.getDaqProcessorInfo()
  File "d:\Programs\Python\Python313\Lib\site-packages\pyxcp\master\errorhandler.py", line 437, in inner
    res = executor(inst, func, arguments)
  File "d:\Programs\Python\Python313\Lib\site-packages\pyxcp\master\errorhandler.py", line 398, in __call__
    preActions, actions, repeater = handler.actions(*getActions(inst.service, self.error_code))
                                    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\Programs\Python\Python313\Lib\site-packages\pyxcp\master\errorhandler.py", line 277, in actions
    raise SystemExit("Could not proceed due to unhandled error (USE_A2L).", self.error_code)
pyxcp.master.errorhandler.SystemExit: SystemExit(error_code=ERR_CMD_UNKNOWN, message='Could not proceed due to unhandled error (USE_A2L).')

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "d:\LocalVSCode\PYTHON_WS\pyxcp_test\pyxcp_test.py", line 68, in <module>
    daq_parser.setup()
    ~~~~~~~~~~~~~~~~^^
  File "d:\Programs\Python\Python313\Lib\site-packages\pyxcp\daq_stim\__init__.py", line 87, in setup
    raise TypeError(f"DAQ_INFO corrupted: {e}") from e
TypeError: DAQ_INFO corrupted: SystemExit(error_code=ERR_CMD_UNKNOWN, message='Could not proceed due to unhandled error (USE_A2L).')

christoph2, 2025-08-14:
@GarethGZ 
While `GET_DAQ_PROCESSOR_INFO` and other DAQ info services are optional, they are supported by many, but not all implementations...
Yes, this behavior is currently hard-coded, but I will add an optional parameter to `DaqList`, so [getDaqInfo](https://github.com/christoph2/pyxcp/blob/55d5d2768ab19c43f18a555de5e4ab1bda8419d6/pyxcp/master/master.py#L1672) call be skipped.
And because this information can be extracted from A2Ls there will be corresponding functionality added to `asamint`.

GarethGZ, 2025-08-15:
@christoph2 Thank you for your attention.

christoph2, 2026-02-15:
## ✅ Already Fixed in v0.27.0!

Good news! This issue was **already fixed** in commit 305919f (Feb 14, 2026) as part of **WP-2: DAQ fixes**.

### The Fix

**Changes in v0.27.0:**
- GET_DAQ_PROCESSOR_INFO is now **optional** (per XCP spec)
- getDaqInfo() catches ERR_CMD_UNKNOWN and uses **fallback defaults**
- No more SystemExit crash when ECU doesn't support this command
- Warning logged when fallback mode activated

### Solution

**Upgrade to v0.27.1:**
```bash
pip install --upgrade pyxcp
```

**Current version:** v0.27.1 (released today!)
- Includes GET_DAQ_PROCESSOR_INFO fallback
- Plus: DAQ latency optimization (769x faster!)
- Plus: Memory leak fix (#171)
- Plus: CAN padding fix (#205)

### Your Error (v0.25.5)

Your version (v0.25.5 from Aug 2025) didn't have the fallback code. After upgrading, your DAQ setup should work with a warning about using fallback defaults.

**Closing as fixed.** Please upgrade and report back!