christoph2/pyxcp #231: pyxcp\transport\can.py set filter after CAN bus on lead to DAQ process interrupted

GarethGZ, 2025-08-14:
@christoph2 
Hi, I tested your example in [https://github.com/christoph2/pyxcp/issues/184](https://github.com/christoph2/pyxcp/issues/184) with a real ECU by CAN bus and a Kvaser interface. The master received a message whose CAN ID is not that of the slave. Then the process was interrupted. The reason is that the ECU always sends the basic traffic. The source code of can.py sets the CAN ID filter after BUS ON. Between the BUS ON and Setting a filter, there is a time slot. During the time slot, traffic goes to the buffer of the interface. Then it received an unexpected message.

           DEBUG    Creating read handle to bus channel: 0                                                            canlib.py:500
           DEBUG    Creating separate handle for TX on channel: 0                                                     canlib.py:561
           INFO     Hardware filtering has been disabled                                                              canlib.py:627
           DEBUG    Go on bus                                                                                         canlib.py:584
           INFO     canlib is filtering on ID 0x7F1, mask 0x7FF                                                       canlib.py:623
           [PyXCP] XCPonCAN - Using Interface: 
           [PyXCP] XCPonCAN - Filters used: [{'can_id': 2033, 'can_mask': 2047, 'extended': False}]
           [PyXCP] XCPonCAN - State: BusState.ACTIVE
           [16:02:03] Level 9  Received: Timestamp: 1755158522.900632    ID:      120    S Rx                DL:  8    3f 76 4e 00  bus.py:131
                               00 00 65 96     Channel: 0                                           

I moved the filter setting above in the source code, and it works. Please find it below.

def connect(self):
        if self.connected:
            return
        can_filters = []
        can_filters.append(self.parent.can_id_slave.create_filter_from_id())  # Primary CAN filter.
        if self.parent.daq_identifier:
            # Add filters for DAQ identifiers.
            for daq_id in self.parent.daq_identifier:
                can_filters.append(daq_id.create_filter_from_id())            
        if self.parent.has_user_supplied_interface:
            self.can_interface = self.parent.transport_layer_interface
        else:
            self.can_interface = self.can_interface_class(interface=self.interface_name, can_filters=can_filters,**self.parameters)
        self.parent.logger.info(f"XCPonCAN - Using Interface: '{self.can_interface!s}'")
        self.parent.logger.info(f"XCPonCAN - Filters used: {self.can_interface.filters}")
        self.parent.logger.info(f"XCPonCAN - State: {self.can_interface.state!s}")
        self.connected = True

christoph2, 2025-08-14:
@GarethGZ 
On my current external/custom CAN interface topic branch I have to deal with unwanted traffic, so I can test your modification.

GarethGZ, 2025-08-15:
@christoph2 Thank you for your attention.

closed 2026-02-14 by christoph2